espressif/arduino-esp32 #9079: Supporth for linkStatus() to detect Ethernet cable when using w5500

kapyaar, 2024-01-07:
### Related area

ESP32 + ETH

### Hardware specification

ESP32 Dev Module, Generic with W5500

### Is your feature request related to a problem?

In the Ethernet library for W5500, there is option to detect if ethernet cable is plugged in. 

https://github.com/arduino-libraries/Ethernet/blob/39103da0e1bc569023625ee4693272773397dbb6/src/Ethernet.cpp#L98

Would be nice to have this available when using ETH.h 

### Describe the solution you'd like

Just like linkUP(), a linkStatus() function would suffice. 

### Additional context

The purpose is to decide if cable is present, and if not, ditch Ethernet and switch to WiFi. 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-01-08:
@kapyaar since we are using ESP-IDF's ETH driver, that request should be best posted there.

cc @kostaond

kostaond, 2024-01-08:
@me-no-dev, link up is already indicated by the Ethernet driver by raising `ETHERNET_EVENT_CONNECTED` event. [Here](https://github.com/espressif/esp-idf/blob/release/v5.1/examples/ethernet/basic/main/ethernet_example_main.c#L31) is example how to use it. Therefore, I think there is no need for ESP-IDF's Ethernet driver.

kapyaar, 2024-01-08:
@me-no-dev ok, got it :) I was not sure if this feature was there, but not exposed in code. Closing this. 

@kostaond On another post, you had mentioned I try esp32.com, 
but that page does not load for any searches. Tried many times, different browsers, w and w/o vpn, etc. See screenshots for Brave and firefox. 

[image]

kapyaar, 2024-01-08:
@kostaond ETHERNET_EVENT_CONNECTED (if same as ARDUINO_EVENT_ETH_CONNECTED) probably only checks if w5500 is there or not? I tested without the cable plugged in, and it still prints "ETH Started". 

On the otherhand, 

 if (Ethernet.linkStatus() == LinkOFF) {
    Serial.println("Ethernet cable is not connected.");
  }

is specifcally looking for cable.

me-no-dev, 2024-01-08:
@kapyaar 
- STARTED : Driver has started and connected to the chip
- CONNECTED : Cable is plugged in and there is a device on the other end (link up)
- GOT_IP : IP address has been obtained from DHCP
- DISCONNECTED : Cable was unplugged on either end (link down)
- STOPPED : Driver has been uninitialized and resources freed

kapyaar, 2024-01-08:
@me-no-dev 

Understood. That is all fine. But consider the following situations. 
1. Ethernet cable is plugged in, Driver has started, but the DHCP server is taking some time (or fails) to assign an IP.
2. Ethernet cable is unplugged. Driver starts, but never reaches "Connected" or further.

In both cases, We are waiting at "STARTED". Never goes beyond that. And Can't say conclusively if it is 1 or 2. If only Ethernet is available, alright, let's leave it there. But Since WiFi is an option here, resolving the above can be helpful, to switch to wifi as and when. One could use a timeout, but given the functionality is present, making it available would be helpful. That's all.

kostaond, 2024-01-09:
Ad 1) You can isolate such situation by receiving `ETHERNET_EVENT_START` and `ETHERNET_EVENT_CONNECTED` but you don't get `IP_EVENT_ETH_GOT_IP`.
Ad 2) You just get `ETHERNET_EVENT_START`.

me-no-dev, 2024-01-09:
@kapyaar we will add a way to ask if link is up, but it will be done the same exact way: on `CONNECTED` set link `UP` and on `DISCONNECTED` set link `DOWN`. You can do this yourself by listening to the events. This is true for `STA` also. `GOT_IP` is sent only when static IP was not set and DHCP server responded. This has nothing to do with the physical layer (`CONNECTED`/`DISCONNECTED`)

closed 2025-01-15 by Parsaabasi