espressif/arduino-esp32 #9082: esp32c3 SPI SD can't not work

TasLR, 2024-01-09:
### Board

esp32-c3-wroom-02

### Device Description

esp32-c3-wroom-02-N4

### Hardware Configuration

#define PIN_NUM_MISO  10
#define PIN_NUM_MOSI  19
#define PIN_NUM_CLK   18
#define PIN_NUM_CS    3

### Version

v2.0.14

### IDE Name

platformIO

### Operating System

windows10

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

I want to run spi sd test with esp32c3, but It can't work under arduino framework,and It can work under the esp idf framework. I use the same ports on these example. under arduino framework, the log is sd drive fail, can you tell how to do slove this problem? thank you !

### Sketch

```cpp
/*
 * pin 1 - not used          |  Micro SD card     |
 * pin 2 - CS (SS)           |                   /
 * pin 3 - DI (MOSI)         |                  |__
 * pin 4 - VDD (3.3V)        |                    |
 * pin 5 - SCK (SCLK)        | 8 7 6 5 4 3 2 1   /
 * pin 6 - VSS (GND)         | ▄ ▄ ▄ ▄ ▄ ▄ ▄ ▄  /
 * pin 7 - DO (MISO)         | ? ? █ ? █ ? ? ? |
 * pin 8 - not used          |_________________|
 *                             ║ ║ ║ ║ ║ ║ ║ ║
 *                     ╔═══════╝ ║ ║ ║ ║ ║ ║ ╚═════════╗
 *                     ║         ║ ║ ║ ║ ║ ╚══════╗    ║
 *                     ║   ╔═════╝ ║ ║ ║ ╚═════╗  ║    ║
 * Connections for     ║   ║   ╔═══╩═║═║═══╗   ║  ║    ║
 * full-sized          ║   ║   ║   ╔═╝ ║   ║   ║  ║    ║
 * SD card             ║   ║   ║   ║   ║   ║   ║  ║    ║
 * Pin name         |  -  DO  VSS SCK VDD VSS DI CS    -  |
 * SD pin number    |  8   7   6   5   4   3   2   1   9 /
 *                  |                                  █/
 *                  |__▍___▊___█___█___█___█___█___█___/
 *
 * Note:  The SPI pins can be manually configured by using `SPI.begin(sck, miso, mosi, cs).`
 *        Alternatively, you can change the CS pin and use the other default settings by using `SD.begin(cs)`.
 *
[... 241 lines cut ...]

}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
[   394][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[   394][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   698][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Card Mount Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-01-09:
please post your PIO and Arduino IDE board configuration

TasLR, 2024-01-09:
> please post your PIO and Arduino IDE board configuration

I use default pio board configuration

TasLR, 2024-01-09:
I select ESP32-C3-DevKitC-02 board to create the project

SuGlider, 2024-01-09:
PIO default means Arduino Core 2.0.10 and IDF 4.4.5

SuGlider, 2024-01-09:
@TasLR - Please chnage your Platformi.ini to use Arduino Core 2.0.14:

```
[env:esp32-c3-devkitc-02]
platform = espressif32 @ ^6.5.0  ; based on Arduino Core 2.0.14
board = esp32-c3-devkitc-02
framework = arduino
```

Test the sketch again and let us know.

SuGlider, 2024-01-09:
Building the sketch using PIO 6.5.0 should report:

```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|-- SPI @ 2.0.0
```

Jason2866, 2024-01-09:
@SuGlider 
PIO default means Arduino Core 2.0.14

to be sure to use latest, it is recommended to pin the version
So latest is used with:
```
platform = espressif32 @ 6.5.0
```

if the version is left out any version can be used which is "found" (formerly installed)

SuGlider, 2024-01-09:
@Jason2866 - Got the 2.0.10 out of a fresh new project. Using just the `platform = espressif32`.  
The only way to make it install 2.0.14 was by setting the version with `platform = espressif32 @ 6.5.0`.

Is this how it should work, or it may happen because I installed PIO for the first time quite a time ago?

Jason2866, 2024-01-09:
> It may happen because I installed PIO for the first time quite a time ago?

Exactly this. If you nuke Platformio with all hidden folders and use `platform = espressif32 `
it will install 6.5.0

TasLR, 2024-01-10:
> @TasLR - Please change your Platformi.ini to use Arduino Core 2.0.14:
> 
> ```
> [env:esp32-c3-devkitc-02]
> platform = espressif32 @ ^6.5.0  ; based on Arduino Core 2.0.14
> board = esp32-c3-devkitc-02
> framework = arduino
> ```
> 
> Test the sketch again and let us know.

I went to configure my project file according to the configuration you showed, but I still have the same problem，the build log as following:
  PACKAGES:
   - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
   - tool-esptoolpy @ 1.40501.0 (4.5.1)
   - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
  LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 34 compatible libraries
  Scanning dependencies...
  Dependency Graph
  |-- FS @ 2.0.0
  |-- SD @ 2.0.0
  |-- SPI @ 2.0.0
  Building in release mode

TasLR, 2024-03-26:
> @TasLR - Please change your Platformi.ini to use Arduino Core 2.0.14:
> 
> ```
> [env:esp32-c3-devkitc-02]
> platform = espressif32 @ ^6.5.0  ; based on Arduino Core 2.0.14
> board = esp32-c3-devkitc-02
> framework = arduino
> ```
> 
> Test the sketch again and let us know.

I've reconfigured the project the way you said, but I'm still having the same problem, could you please test it out

lbernstone, 2024-03-26:
There's a spot in setup where you need to change the code to reassign pins.  I'll submit a PR to fix the example.

TasLR, 2024-03-27:
> There's a spot in setup where you need to change the code to reassign pins. I'll submit a PR to fix the example.

Isn't the SPI pin of ESP32 series microcontroller mapped to any pin? Because I can drive normally according to the pin test you use, but I can't change to other pins, is this the difference in the program framework or some other reason

commit 654aeada (2024-03-27): Matched pin numbers in doco to numbers in variants files (#9422)

* Matched pin numbers in doco to numbers in variants files

* Non-breaking dashes

* Changed SD examples to make REASSIGN_PINS more transparent. Fixes #9082

---------
closed 2024-03-27 by me-no-dev