espressif/arduino-esp32 #9087: ESP32-S3 with Arduino IDE

mickgillen, 2024-01-10:
### Board

ESP32-S3 Micro Processor

### Device Description

[image]
[image]

### Hardware Configuration

there is no development board, the IC is fitted to a new PCB product!!

### Version

other

### IDE Name

Arduino

### Operating System

windows 10

### Flash frequency

240Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

we are trying to run a ESP32-S3 (exact part is ESP32-S3-Wroom-2-N32R8V, Digikey 1965-ESP32-S3-WROOM-2-N32R8VCT-ND) microprocessor on a PCB with IDE Arduino, which program should be selected from the board manager in the Arduino IDE?

Using ESP32S3 Dev Module from the menu list in the Arduino IDE, it gives an error AFTER download has completed correctly:

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x403776f6:0x3fceb180 0x4037a8b5:0x3fceb1a0 0x4038004d:0x3fceb1c0 0x4200b796:0x3fceb2f0 0x40377067:0x3fceb320 0x403cd823:0x3fceb350 0x403cdae2:0x3fceb380 0x403c9929:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

ELF file SHA256: 845b972cfb84bccc

E (187) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377304
SPIWP:0xee

we have been testing a ESP32 DEV KIT board which works fine with ESP32 DEV KIT from the Arduino board manager menu list!!!

### Sketch

```cpp
our code is not the issue, we need to know what board manager code will run the ESP32-S3 micro which is attached to our PCB!!
```

### Debug Message

```plain
assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x403776f6:0x3fceb180 0x4037a8b5:0x3fceb1a0 0x4038004d:0x3fceb1c0 0x4200b796:0x3fceb2f0 0x40377067:0x3fceb320 0x403cd823:0x3fceb350 0x403cdae2:0x3fceb380 0x403c9929:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

ELF file SHA256: 845b972cfb84bccc

E (187) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377304
SPIWP:0xee
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Rainbow-Cai, 2024-01-10:
Hi @mickgillen  When use the [ESP32-S3-WROOM-2](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf) module ， please follow the below screenshot to set the software. Because the ESP32-S3-WROOM-2 is used the Octal Flash and Octal PSRAM.

[image]

SuGlider, 2024-01-10:
Thanks @Rainbow-Cai. This is it.
@mickgillen - please use the configuration above.
Close the issue in case that it has been solved.

VojtechBartoska, 2024-02-12:
@mickgillen is this still valid?

lbernstone, 2024-02-12:
In 3.0 there is also an entry specifically for ESP32-S3-WROOM-2.

closed 2024-09-10 by VojtechBartoska