espressif/arduino-esp32 #9088: Nullpointer check before free() or delete()

everslick, 2024-01-10:
### Board

n.a

### Device Description

n.a.

### Hardware Configuration

n.a.

### Version

latest master (checkout manually)

### IDE Name

n.a.

### Operating System

n.a.

### Flash frequency

n.a.

### PSRAM enabled

yes

### Upload speed

n.a.

### Description

What is the official stance on the numerous places where pointers are checked for NULL before free() or delete()? Should we get rid of them?

I'm especially curious about some code like this:

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiUdp.cpp#L108-L112

I believe `tx_buffer` is first assigned to `b` here to make it somehow **thread-save**, but is this really something that gives any guarantees and is this a consistent pattern throughout the whole codebase? (I do not think so, the `tx_buffer` above is not given the same treatment.) IMHO the code should be reduced to:

```
  delete rx_buffer;
  rx_buffer = NULL;
```

What would be the preferred way of submitting PRs here? file by file / lib by lib? Anything else?

### Sketch

```cpp
n.a.
```

### Debug Message

```plain
n.a.
```

### Other Steps to Reproduce

n.a.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2024-01-12:
@me-no-dev Do you know why this part of the `WiFiUdp` was implemented like this ?

me-no-dev, 2024-01-12:
the whole implementation needs to be changed here to use FreeRTOS storage (Queue/RingBuffer). It's not safe at all.

everslick, 2024-01-15:
OK, this still leaves the question, if we should purge the check for NULL before free() or leave it alone (in regard to the whole codebase, not only WiFiUDP). I believe the check is so common, because so many people do not know, that it is OK to delete/free NULL pointers and thus feel the need to guard it. but IMHO, code that has no purpose should be eliminated.

VojtechBartoska, 2024-05-23:
@me-no-dev was this covered by Network refactoring?

me-no-dev, 2024-05-23:
@everslick while we can free NULL pointers, for readability and clarity it makes sense to me to leave the checks. Also good idea to be used to check for when using other "free-like" functions in the SDK (for freeing scan results, etc.). All this is my personal opinion

everslick, 2024-05-23:
Thanks for making this clear. I will close this issue as answered then. :)