espressif/arduino-esp32 #9090: ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution.

denck, 2024-01-11:
### Board

ESP32 Dev Module

### Device Description

ESP32-DevKitC-32E on a custom PCB board

### Hardware Configuration

ledcSetup(0, 1E5, 12);
ledcAttachPin(25, 0)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

MacOS

### Flash frequency

not sure

### PSRAM enabled

yes

### Upload speed

115200

### Description

So we have a product that uses a ESP32-DevKitC-32E. We already shipped out around 150 products for a few months now. Now I got one that is not working and if I check the serial monitor in Arduino IDE, I get this error message:
E (28) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=50

I think it has something to do with the buzzer that is connected, but I'm not really sure.

### Sketch

```cpp
The code is too long to put here.
```

### Debug Message

```plain
E (28) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=50
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

denck, 2024-01-11:
I can provide gerber files if somebody is willing to see how everything is connected and what could cause this error in this particular case. All the others are running fine with the exact same setup and software.

VojtechBartoska, 2024-01-11:
@P-R-O-C-H-Y please help with triage, thanks!

denck, 2024-01-11:
Just found that if I put the two lines below in comment and upload the software the product works like it should.
Only thing is that the buzzer doesn't work anymore. So it  has something to do with the buzzer, sure about that.
//ledcSetup(0, 1E5, 12);
//ledcAttachPin(25, 0)

P-R-O-C-H-Y, 2024-01-15:
Hi @denck, you can run [this example](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogOut/ledcFrequency/ledcFrequency.ino) to see what ranges of frequencies and resolutions are valid.

TD-er, 2024-02-04:
I think there is another issue here.
I was about to open a new issue about this, but since the template asks about me searching for similar issues... ;)

I think the problem here might be that `ledc_handle.used_channels` never gets initialized to 0.

Not sure if it is the best fix, but a really quick one for testing could be to add this to the start of this function:
```c++
bool ledcAttach(uint8_t pin, uint32_t freq, uint8_t resolution)
{
    if (!fade_initialized) {
        ledc_handle.used_channels = 0;
    }
...
```

TD-er, 2024-02-04:
Oh and if this issue is actually about the resolution/duty cycle, then maybe this code of mine can help here.
It is a piece of code I use to compute duty cycle, frequency and resolution to fade from one to another.
```c++
    uint8_t  resolution  = 10;
    uint32_t start_duty  = 0;
    uint32_t target_duty = dutyCycle;

    if (/*(tempStatus.mode == PIN_MODE_PWM) && */ (ledcReadFreq(gpio) != 0)) {
      start_duty = ledcRead(gpio);
    } else {
      // No PWM active, pick proper estimated start_duty based on pin state
      if (digitalRead(gpio)) {
        start_duty = (1 << resolution);
      }
    }

    if (frequency > 0) {
      // Adjust the frequency and resolution to keep them in the range of the used timer frequency.
      while ((40000000u >> resolution) < frequency && resolution > 7) {
        --resolution;

        if (start_duty > target_duty) {
          ++start_duty;
        }
        else {
          ++target_duty;
        }
        start_duty  >>= 1;
        target_duty >>= 1;
      }
    }
    start_duty  = std::min(start_duty, static_cast<uint32_t>((1 << resolution) - 1));
    target_duty = std::min(target_duty, static_cast<uint32_t>((1 << resolution) - 1));
```

VojtechBartoska, 2024-02-12:
cc @P-R-O-C-H-Y

P-R-O-C-H-Y, 2024-02-13:
@denck Do you still need any help? I am closing this one marking as solved. Feel free to reopen/comment anytime if you encounter any problem.

The possible problem with the `ledc_handle.used_channels` never gets initialized to 0 has separate issue opened.

closed 2024-02-13 by P-R-O-C-H-Y