espressif/arduino-esp32 #9092: Support for ESP32-C61 (with PSRAM)

TD-er, 2024-01-11:
### Related area

SoC support

### Hardware specification

Support for ESP32-C61

### Is your feature request related to a problem?

As announced new SoC by Espressif: https://www.espressif.com/en/news/ESP32-C61_SoC

### Describe the solution you'd like

Please add support for the PSRAM in the ESP32-C61 (as of now the only in the C-series with PSRAM)

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-01-12:
- Chip and modules must be first produced
- Support needs to be added to ESP-IDF
- Arduino needs to start tracking the IDF version that has support

When the above becomes reality, we can add support for it :)

lbernstone, 2024-01-12:
Gimme!

TD-er, 2024-01-12:
> When the above becomes reality, we can add support for it :)

Hmm why can't we do "DDD" (Demand Driven Development), so we implement support for a not even announced chip and Espressif will follow our implementation :)

J/K of course.
I think 2024 will be a busy year for implementing support of all already announced products but with no silicon yet (ESP32-C5 / P4 / C61 / H2 
Now with the well structured ESP-IDF 5.x it has become a lot simpler to implement support for a new platform.

lbernstone, 2024-03-11:
P4 is going to be very complicated to implement.  I suspect C61 may be able to use the exact same methodology as WROVER.

VojtechBartoska, 2024-03-12:
Hello @TD-er,

thanks for opening a ticket, you are fast! :)

We will start first with ESP32-P4, it will be fully supported by ESP-IDF v5.3, which will be released at the middle of this year. We will soon open an issue to track progress on ESP32-P4.

ESP32-C5 & ESP32-C61 will have preview support in ESP-IDF v5.3 and full support in v5.4 so there is still time to start working on it and we do not have HW yet.

I hope it will answer your questions.

I have assigned this task to 3.2.0 milestone.

TD-er, 2025-06-30:
Just a small update on this...
I just received two ESP32-C61 boards, I bought myself from the Espressif Official Store on AliExpress.
I know there is no support jet in Arduino for it and I also need to wait for ESP-IDF5.5.1.

But I wonder if it is possible to have at least _some_ docs released as there is literally nothing right now other than how to flash it and in what kind of karton the reels of parts will be shipped in to manufacturers.
Since I'm implementing MCU specifics for the C5 and P4 to ESPEasy, it would be nice if I can also do the same for the C61 while I'm working in that part of the code already.
For example, even which GPIO pins are usable or is a boot strapping pin, isn't made public.
Preparing these things now will take less time compared to when I have to search for it again when the documentation becomes available.

TD-er, 2025-07-19:
Finally there are [some docs](https://www.espressif.com/en/support/documents/technical-documents?keys=&field_type_tid_parent=esp32c61Series-SoCs&field_type_tid%5B%5D=1637&field_type_tid_parent=esp32c61Series-Modules&field_type_tid%5B%5D=2601&field_type_tid_parent=esp32c61Series-DevKits&field_type_tid%5B%5D=2603) released.

For example [ESP32C61 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c61-wroom-1_wroom-1u_datasheet_en.pdf)