espressif/arduino-esp32 #9096: esp32 failed to upload the code

Mr-Essid, 2024-01-12:
### Board

esp32 wroom

### Device Description

no other hardware attached to it

### Hardware Configuration

no wires

### Version

latest master (checkout manually)

### IDE Name

esptool

### Operating System

linux

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

A fatal error occurred: Packet content transfer stopped (received 7220 bytes)
in other hand my board keep printing 0xffffffff invalid header in the serial

### Sketch

```cpp
just erase flash / no sketch
```

### Debug Message

```plain
esptool.py erase_flash 
esptool.py vv4.8.dev1
Found 2 serial ports
Serial port /dev/ttyUSB0 // note that the esp in this port
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:87:ce:2f:28:64
Uploading stub...
Running stub...
Stub running...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-01-12:
Does `./esptool.py flash_id` give you the flash info?

UserNameUTF-8, 2024-01-12:
> Does `./esptool.py flash_id` give you the flash info?

The default command esptool.py * (erase_flash, write_flash ..) PORT have 
The right info I have tried before and it works fine

Jason2866, 2024-01-13:
Try latest stable release esptool.py (v4.7). Try different USB cable. Which serial USB Adapter used? Really bad are FTDI ones. Can't deliver enough current for flashing.

UserNameUTF-8, 2024-01-13:
> Try latest stable release esptool.py (v4.7). Try different USB cable. Which serial USB Adapter used? Really bad are FTDI ones. Can't deliver enough current for flashing.

I have another card and it works fine with the same usb cable and config that's why might be the card broken

Jason2866, 2024-01-13:
So, you can close this. No bug in Arduino Core.

SuGlider, 2024-01-13:
>A fatal error occurred: Packet content transfer stopped (received 7220 bytes)

Upload failed. It can be due to bad cable, bad connectors, bad Ground contact, flash failure, problems with the CP2102 (USB-UART onboard chip).

>in other hand my board keep printing 0xffffffff invalid header in the serial

This happens because there is no valid firmware in the flash.

Mr-Essid, 2024-01-13:
> Upload failed. It can be due to bad cable, bad connectors, bad Ground contact, flash failure, problems with the CP2102 (USB-UART onboard chip).
>
>
> This happens because there is no valid firmware in [...]

please how can remove the invalid firmware from the flash

Jason2866, 2024-01-13:
See here https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html

Mr-Essid, 2024-01-14:
broken chip

closed 2024-01-14 by Mr-Essid