espressif/arduino-esp32 #9098: SD_MMC strange behavior opening file for read and write

daniel-dona, 2024-01-13:
### Board

AI Thinker ESP32-CAM

### Device Description

AI Thinker ESP32-CAM with micro SD card

### Hardware Configuration

None

### Version

v2.0.14

### IDE Name

VSCode + PlatformIO

### Operating System

Ubuntu 22.04

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I was trying to open a file to read a counter and update it incrementing the counter and found some strange behavior.

Opening a file of the SD card this way works as expected, file content is one line just with "000000000001":

``` c
File file = SD_MMC.open(path);
```

I'm able to read the file contents normally.

```
[    20][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   469][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   490][I][main.cpp:16] setup(): Initializing SD
[   548][I][main.cpp:24] setup(): Initialized correctly
[   553][I][main.cpp:31] setup(): File size: 13
[   554][I][main.cpp:33] setup(): Read from file: 
000000000001
[   555][I][main.cpp:41] setup(): Reading done
```

But opening the file this way, I'm unable to read the file contents.

``` c
File file = SD_MMC.open(path, FILE_APPEND);
```

```
[    20][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   469][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   490][I][main.cpp:16] setup(): Initializing SD
[   548][I][main.cpp:24] setup(): Initialized correctly
[   554][I][main.cpp:31] setup(): File size: 13
[   554][I][main.cpp:33] setup(): Read from file: 
[   554][I][main.cpp:41] setup(): Reading done
```

And finally, opening the file this way results on the contents being wiped.

```c
File file = SD_MMC.open("/metadata/index.txt", FILE_WRITE);
```

```
[    20][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   469][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   490][I][main.cpp:16] setup(): Initializing SD
[   548][I][main.cpp:24] setup(): Initialized correctly
[   557][I][main.cpp:31] setup(): File size: 13
[   557][I][main.cpp:33] setup(): Read from file: 
�����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
```

And the reading loop never ends.

### Sketch

```cpp
#include <Arduino.h>

#include <FS.h>
#include <SD_MMC.h>

void setup() {

  esp_log_level_set("*", ESP_LOG_VERBOSE);

  // LOG UART

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  log_i("Initializing SD");

  if (!SD_MMC.begin("/sdcard", true)){

      log_e("Initialization of SD Card Failed");

  }else{

    log_i("Initialized correctly");

    if(SD_MMC.exists("/metadata/index.txt")){

      //File file = SD_MMC.open("/metadata/index.txt");
      File file = SD_MMC.open("/metadata/index.txt", FILE_WRITE);

      log_i("File size: %d", file.size());

      log_i("Read from file: ");

      while(file.available()){

        Serial.write(file.read());

      }

      log_i("Reading done");

    }

  }

}

void loop() {

}
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

Steps to reproduce:

- Freshly formatted SD card.
- Create the testing file using a computes and safely remove the SD to ensure no data corruption.
- Execute the minimal testing code on ESP32 with the card inserted.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

daniel-dona, 2024-01-13:
The platformio.ini file contents:

```
[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
monitor_speed = 115200

build_flags =
    -O2
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    -D LED_BUILTIN=4
    -D BOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
```

Build log:

```
Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (6.5.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FS @ 2.0.0
|-- SD_MMC @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/esp32cam/firmware.elf
Checking size .pio/build/esp32cam/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.8% (used 22356 bytes from 327680 bytes)
Flash: [=         ]  11.5% (used 362121 bytes from 3145728 bytes)
============================================ [SUCCESS] Took 1.04 seconds ============================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

lbernstone, 2024-01-14:
This is standard posix behavior.  https://man7.org/linux/man-pages/man3/fopen.3.html
You can use the seek() function if you want to change the file position.

daniel-dona, 2024-01-14:
> This is standard posix behavior. https://man7.org/linux/man-pages/man3/fopen.3.html You can use the seek() function if you want to change the file position.

The real problem is `File file = SD_MMC.open("/metadata/index.txt", FILE_WRITE);` wiping file contents and reporting `file.available()` always true.

lbernstone, 2024-01-14:
[This](https://github.com/espressif/arduino-esp32/blob/master/libraries/FS/src/FS.cpp#L59) is what available() does.  Note that until you commit (flush/close) the change made by trunc'ing the file, the file size is still > 0.

VojtechBartoska, 2024-05-23:
Closing as it was already answered.

closed 2024-05-23 by VojtechBartoska