espressif/arduino-esp32 #9110: WiFi on AP mode stopped working (forever?) after softAPdisconnect()

acautomacaoecontrole, 2024-01-15:
### Board

Generic development board with a ESP-WROOM-32

### Device Description

Brandless generic Development board and ESP32 board witj ESP32 module adapter.

### Hardware Configuration

Not aplicable

### Version

latest master (checkout manually)

### IDE Name

PlatformIO and Arduino

### Operating System

Widows 8.1

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi!

I've been working on a Wi-Fi AP project and doing several changes to experiment results and after using softAPdisconnect(), I've noticed ESP32 was not transmitting anymore, even after resetting, turning it off and on and erasing flash.
At a initial moment, I could not realize that this problem happened after using softAPdisconnect().

I've removed little by little the last changes but nothing made the AP work again.

So, I've taken a second ESP32 module and the AP project came back to life. I spared the first ESP32 for a later investigation.

Back to the development of the AP project, at certain moment I included again softAPdisconnect() and the previous problem arose again. This time I could realize that it was related to this function, but I have no clue why, even after checking WiFiAP.cpp to understand what this fucttion does.

Again, a new ESP32 module was taken and the AP functions again. This time softAPdisconnect() was removed.

I do not know if what caused the problem was either softAPdisconnect( true ) or softAPdisconnect( false ). Both?

I've spending A LOT of time trying to recover those 2 ESP32 modules in order to be used as AP again. All failed till now. As a station (client), it works.

BTW, Functions "enableSTA(true)", "WiFi.mode( WIFI_MODE_STA )", "esp_wifi_set_mode( mode )" ("mode" previously set to "WIFI_MODE_STA") do not change mode value. It's always "WIFI_MODE_NULL", according to the result of "esp_wifi_get_mode( &mode )". It ALWAYS returns "ESP_OK" and put the "WIFI_MODE_NULL" value on "mode" variable.
Notice: this constant return of "WIFI_MODE_NULL" happens even on a ESP32 module that is AP operative.

So, the questions:
1) How I will make those two ESP32 modules back to normal?
2) What should I do to stop ESP32 of working as AP but without preventing it from work as AP again?

### Sketch

```cpp
#include  <Arduino.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>
#include "esp_wifi.h"
[... 396 lines cut ...]
    if(!softap_config_equal(conf, conf_current)) {
    	err = esp_wifi_set_config((wifi_interface_t)WIFI_IF_AP, &conf);
        if(err){
        	log_e("set AP config failed");
ets_printf( "\n=5\n" );//ssrxxxteste            
            return false;
        }
    }
//ssrxxxteste
ets_printf( "\nsoftAP() - =6\n"
            "\tssid: %s\n"
            "\tpass: %s\n"
            "\tcana: %d\n"
            "\thidd: %d\n"
            "\tmac_: %d\n"
            "\tstm_: %d\n",
            ssid, passphrase, channel, ssid_hidden, max_connection, ftm_responder );
//ssrxxxteste
    return true;
}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-01-16:
where in the example sketch do I call `WiFi.softAPdisconnect()`? After printing IP, after some time? On some event?

VojtechBartoska, 2024-02-12:
any updates @acautomacaoecontrole?

kestasli, 2024-02-12:
I got exactly same problem with Lilygo esp32s3 T-display. Not sure if WiFi.softAPdisconnect() was executed, as I was using WiFiMmanager library. But now it is not possible to run module in AP mode no matter what (2x modules). Third module (which was not used during the project) runs totally OK.

VojtechBartoska, 2024-05-23:
Hello, networking was refactored for 3.0.0 version. Will you be able to retest in on newest version?

sle118, 2024-06-09:
I am experiencing the same issue on Arduino 3.0.1. 

The device has been working just fine, using the wifimanager library and the arduino framework 2+, up until the point where I tried V3.0.0 and things went south. I went as far as erasing flash and flashing the WiFiAccessPoint example with no success. Logs don't show any error either. 

```
  Arduino Version   : 3.0.1
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=verbose,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
============ Before Setup End ============
[   716][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x401640c8
[   728][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 2 successfully set to type GPIO (1) with bus 0x3
[   739][V][esp32-hal-uart.c:408] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   750][V][esp32-hal-uart.c:497] uartBegin(): UART0 not installed. Starting installation
[   761][V][esp32-hal-uart.c:544] uartBegin(): UART0 initialization done.

Configuring access point...
[   786][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY
AP IP address: 192.168.4.1
[   887][V][AP.cpp:106] _onApEvent(): AP Started
[   893][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 19 - AP_START
[   900][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 19 - AP_START
Server started
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   344616 B ( 336.5 KB)
  Free Bytes        :   262352 B ( 256.2 KB)
  Allocated Bytes   :    72352 B (  70.7 KB)
  Minimum Free Bytes:   261856 B ( 255.7 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     2 : GPIO
     3 : UART_RX[0]
============ After Setup End =============
```

me-no-dev, 2024-06-10:
@sle118 I can not reproduce this with your exact settings and code. AP working fine on my end.

sle118, 2024-06-10:
@me-no-dev Since I was more familiar with the esp-idf framework than the Arduino framework (which controls a bit too much of the underlying sdkconfig to get to the bottom of issues), I went ahead and flashed the esp-idf softAP example which gave me a better clue as to what was going on and I give it to you in hope that this also applies to others hitting the same problem.

```
0x40080400: _init at ??:?

load:0x40080404,len:3876
entry 0x4008064c
W (69) rtc_clk_init: Possibly invalid CONFIG_XTAL_FREQ setting (40MHz). Detected 26 MHz.
��♥���lo�
c∟�lc��oo��
l�$��c�►��s|l�c�o►↕N|�↕☻��N��↕�����NN�NN�►��|ΐN�dN$ΐN���dN�♦"∟�$"��Nn��
�p~�N↕Nn�
�b�n��c��nl
```
I was having this gibberish serial output and I just couldn't figure out until flashing the esp-idf example which clearly showed there was a crystal speed mismatch. Once the crystal speed was set to "autodetect", problem went away and the softAP started working again.

I am unsure how I can select the proper crystal speed on the Arduino platform, but I guess I have to fiddle around to know. Other than that, the esp-idf example is working just fine. These settings are key for my board: 

```
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
```

me-no-dev, 2024-06-10:
@sle118 ouch... bad news. We could not keep the crystal to auto in Arduino for other reasons (I can not recall now, but something crucial stopped working) so we dropped support for 26MHz crystals. I only know of one board that uses it (red sparkfun esp32 board)

VojtechBartoska, 2024-09-10:
@acautomacaoecontrole can this issue be closed?

closed 2025-01-15 by Parsaabasi