espressif/arduino-esp32 #9112: ScanNetworks causes reboot in 2.0.14 but not in 2.0.2

TDuxis, 2024-01-16:
### Board

ESP32 Dev Module

### Device Description

DevKitC V4 Wroom

### Hardware Configuration

Nothing is attached to the board

### Version

v2.0.14

### IDE Name

Arduino 2.2.1

### Operating System

Win11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have an application where the ESP is set to AP mode and a device like a mobile phone or laptop connects to the ESP via  an ESPAsyncWebserver web page.
The ESP then lists available WiFi networks in a drop down list and the user can then select a WiFi network that the ESP then connects to in STA mode.

If the web page is reloaded the list of available Wifi networks is refreshed.

The code works in ESP package version 2.0.2 but not after upgrade to 2.0.14. If I revert it works again with 2.0.2

There's quite a lot of code in several files and also an html file upploaded to SPIFFS that calls the scan-function when loaded. I have included a simple sketch that can be used to recreate the problem based on the relevant parts of the code I have. When browsing to http://192.168.4.1/scan the ESP reboots due to the WiFi.scanNetworks() call, apparently causing a watchdog timer to trigger.

I have tried various ways to work around this, but not managed anything, other than reverting to 2.0.2 where this does work.

### Sketch

```cpp
#include <WiFi.h>     
#include <ArduinoJson.h>      
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>  

AsyncWebServer server(80);

void setup() {
  IPAddress apIP(192, 168, 4, 1);

  Serial.begin(115200);
  while(!Serial);
  
  WiFi.scanNetworks();
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("Wifi-test");

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world");
    });

  server.on("/scan", HTTP_GET, [](AsyncWebServerRequest *request) {  // Command to scan for Wifi networks
    String json = "{";

    int n;
    n = WiFi.scanNetworks(false, true);             
        
    ESP_LOGI("INFO", "WiFi ScanComplete: %i networks found", n);

    switch (n) {
      case WIFI_SCAN_FAILED:
        json += "\"status\":-2";
        break;
      case WIFI_SCAN_RUNNING:
        json += "\"status\":-1";
        break;
    }

    if (n >= 0) {
      json += "\"ssid\":[";

      for (int i = 0; i < n; ++i) {
        if (WiFi.SSID(i) != "") {
          if (i) json += ",";
          json += "{";
          json += "\"ssid\":\"" + WiFi.SSID(i) + "\"";
          json += "}";
        }
      }

      json += "],\"status\":";
      json += n;

      WiFi.scanDelete();
    }

    json += "}";
    request->send(200, "application/json", json);
  });
  
  server.onNotFound(onRequest);
  server.begin();
}

int n=0;

void loop() {

  Serial.println(n++);
  if ( n>100) n=0;
  delay(1000);
  
}
```

### Debug Message

```plain
Serial monitor output in verbose mode:

ELF file SHA256: 986bb98ddcb8c52a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[    26][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    86][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   152][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   153][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[  3339][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 14
[  3339][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  3346][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
[  3348][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[  3350][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[  3349][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[  3374][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[  3382][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[  3394][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
0
1
[  4949][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: e8:50:8b:b1:6b:19, AID: 1
[  4949][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[  5275][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[  5276][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
[ 25014][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[ 25014][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
22
23
24
25
26
E (42277) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (42277) task_wdt:  - async_tcp (CPU 0/1)
E (42277) task_wdt: Tasks currently running:
E (42277) task_wdt: CPU 0: IDLE
E (42277) task_wdt: CPU 1: IDLE
E (42277) task_wdt: Aborting.

abort() was called at PC 0x400e5f09 on core 0

Backtrace: 0x40083691:0x3ffbebfc |<-CORRUPTED

ELF file SHA256: 986bb98ddcb8c52a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[    26][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    86][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   152][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   153][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[  3339][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 14
[  3340][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  3346][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
[  3348][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[  3350][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[  3350][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[  3374][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[  3382][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[  3394][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
0
1
2
3
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TDuxis, 2024-01-16:
Here's a log with timestamps

```
01:26:17.952 -> ets Jul 29 2019 12:21:46
01:26:17.952 -> 
01:26:17.952 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
01:26:17.952 -> configsip: 0, SPIWP:0xee
01:26:17.952 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:26:17.952 -> mode:DIO, clock div:1
01:26:17.952 -> load:0x3fff0030,len:1344
01:26:17.952 -> load:0x40078000,len:13964
01:26:17.952 -> load:0x40080400,len:3600
01:26:17.952 -> entry 0x400805f0
01:26:18.233 -> [    26][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
01:26:18.327 -> [    86][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
01:26:18.374 -> [   153][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
01:26:18.374 -> [   154][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
01:26:21.756 -> [  3540][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 17
01:26:21.756 -> [  3540][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 1 - SCAN_DONE
01:26:21.756 -> [  3547][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
01:26:21.756 -> [  3549][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
01:26:21.803 -> [  3551][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
01:26:21.803 -> [  3550][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
01:26:21.803 -> [  3575][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
01:26:21.803 -> [  3583][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
01:26:21.803 -> [  3595][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
01:26:21.849 -> 0
01:26:22.835 -> 1
01:26:23.445 -> [  5205][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: e8:50:8b:b1:6b:19, AID: 1
01:26:23.445 -> [  5206][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
01:26:23.633 -> [  5418][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
01:26:23.633 -> [  5418][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
01:26:23.821 -> 2
01:26:24.806 -> 3
01:26:25.839 -> 4
01:26:26.825 -> 5
01:26:27.810 -> 6
01:26:28.843 -> 7
01:26:29.829 -> 8
01:26:30.815 -> 9
01:26:31.802 -> 10
01:26:32.834 -> 11
01:26:33.820 -> 12
01:26:34.806 -> 13
01:26:35.839 -> 14
01:26:36.824 -> 15
01:26:37.810 -> 16
01:26:38.325 -> [ 20087][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
01:26:38.325 -> [ 20087][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
01:26:38.842 -> 17
01:26:39.828 -> 18
01:26:40.814 -> 19
01:26:41.800 -> 20
01:26:42.833 -> 21
01:26:43.302 -> E (32422) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
01:26:43.302 -> E (32422) task_wdt:  - async_tcp (CPU 0/1)
01:26:43.302 -> E (32422) task_wdt: Tasks currently running:
01:26:43.302 -> E (32422) task_wdt: CPU 0: IDLE
01:26:43.302 -> E (32422) task_wdt: CPU 1: IDLE
01:26:43.302 -> E (32422) task_wdt: Aborting.
01:26:43.302 -> 
01:26:43.302 -> abort() was called at PC 0x400e5f09 on core 0
01:26:43.349 -> 
01:26:43.349 -> 
01:26:43.349 -> Backtrace: 0x40083691:0x3ffbebfc |<-CORRUPTED
01:26:43.349 -> 
01:26:43.349 -> 
01:26:43.349 -> 
01:26:43.349 -> 
01:26:43.349 -> ELF file SHA256: 6b9071df986cea58
01:26:43.349 -> 
01:26:43.489 -> Rebooting...
01:26:43.489 -> ets Jul 29 2019 12:21:46
01:26:43.489 -> 
01:26:43.489 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
01:26:43.537 -> configsip: 0, SPIWP:0xee
01:26:43.537 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:26:43.537 -> mode:DIO, clock div:1
01:26:43.537 -> load:0x3fff0030,len:1344
01:26:43.537 -> load:0x40078000,len:13964
01:26:43.537 -> load:0x40080400,len:3600
01:26:43.537 -> entry 0x400805f0
01:26:43.819 -> [    26][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
01:26:43.866 -> [    86][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
01:26:43.960 -> [   153][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
01:26:43.960 -> [   154][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
01:26:47.152 -> [  3340][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 15
01:26:47.152 -> [  3340][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 1 - SCAN_DONE
01:26:47.152 -> [  3347][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
01:26:47.152 -> [  3349][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
01:26:47.152 -> [  3351][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
01:26:47.152 -> [  3351][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
01:26:47.152 -> [  3375][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
01:26:47.199 -> [  3383][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
01:26:47.199 -> [  3395][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
01:26:47.199 -> 0
01:26:48.184 -> 1
01:26:48.935 -> [  5125][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: e8:50:8b:b1:6b:19, AID: 1
01:26:48.935 -> [  5126][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
01:26:49.170 -> [  5400][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
01:26:49.217 -> [  5401][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
01:26:49.217 -> 2
01:26:50.202 -> 3
01:26:51.188 -> 4

```

Jason2866, 2024-01-16:
`ESPAsyncWebServer` is not updated since a very long time and is known to make issues with newer Arduino versions. Does the issue goes away if you use the Webserver from the examples folder?

TDuxis, 2024-01-16:
> `ESPAsyncWebServer` is not updated since a very long time and is known to make issues with newer Arduino versions. Does the issue go away if you use the Webserver from the examples folder?

Ok, thanks for the heads up!

I notice the examples include WebServer.h, is this library included with the board-package, or which library should I install to get the current?

Jason2866, 2024-01-16:
Worth a try. See example https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples/WebServer

TDuxis, 2024-01-16:
Should I skip AsyncTCP as well?

TDuxis, 2024-01-17:
> Worth a try. See example https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples/WebServer
> 
> We are using `WiFi.ScanNetworks` without issues in project Tasmota. So I doubt there is a general bug there.

Are you using the ESPAsyncWebserver in that project?

I do believe this is a bug, because it sometimes works with minor changes that should make no difference at all.
It is getting late here now, but I will get back with a different SW version that, unexplainably, works.

closed 2025-01-15 by Parsaabasi