espressif/arduino-esp32 #9113: esp 32 -  problems redefining  tx and rx pins on Serial 0  on version 2.0.13 and 2.0.14

xalier1, 2024-01-16:
### Board

LOLIN s2 mini

### Device Description

no other hardware other the  LOLIN s2 mini board required

### Hardware Configuration

A  jumper between the tx and rx pins is needed to show functionality with the sketch provided.

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

Not sure -  not changed from default.

### PSRAM enabled

no

### Upload speed

921600

### Description

Upgrading  from version 2.0.12 to 2.0.14  broke the sketch below until the statement " Serial0.end(true);" was added before the Serial0.begin() statement. The Serial0.end() statement is not required in this sketch if this sketch is run on 2.0.12 or earlier.

### Sketch

```cpp
int count = 0;

void setup() {
         unsigned long start = micros();
         Serial.begin(9600);     // opens serial port, sets data rate to 9600 bps
         while (!Serial  && ( (micros() - start) < 1*1000*1000l)) {
            delay(1);
         }
         if(!Serial) {
          Serial.println("timed out waiting for Serial");
         }
         Serial.println("Starting test");
         //Serial1.begin(9600);
         Serial0.end(true);  // not required by 2.0.12 but required by 2.0.13 and 14
         Serial0.begin(9600, SERIAL_8N1, 37, 39); // rx=37, tx=39 for wemos esp32s2
         // need to jumper pins 37 to 39 to get a uart loopback
         Serial0.println("Counter: "+String(count++)); 
}

void loop() {
  static unsigned long lastTM=millis();  // close enough to send in setup()

  static String x="";
  while (Serial0.available() > 0) {
    // read the incoming byte:
    char incomingByte = Serial0.read();
    x += incomingByte;
  }

  if( millis() - lastTM > 1*1000) {
    lastTM = millis();
    if( x != "" ) {
      Serial.print("Received: "+x);
      x ="";
    }
    else {
      Serial.println("Timed out on receive");
    }    
    Serial0.println("Counter: "+String(count++));
  }
}
```

### Debug Message

```plain
No debug messages go to USB console when core debug level is set to verbose on this LOLIN s2 mini board.
```

### Other Steps to Reproduce

Important to jumper pins 37 to 39 to  connect RX to TX for the serial0 uart to be able to receive what is sent to allow the provided sketch to run  without  receive timeouts messages.

In looking at the  differences between HardwareSerial.cpp for 2.0.12 and 2.0.14, there appear to be changes that assume that serial0 is special and associated with the console.   The  LOLIN-MINI-S2 uses the  "USB CDC on boot:Enabled" setting on the Arduino IDE since there is no  independent chip on the board to translate from an esp uart to the usb for the console.   Changes to the 2.0.14 hardwareserial.cpp code provided a clue to terminate Serial0 before   using  the serial0.begin statement to define the uart rx and tx pins.  The changes in HardwareSerial.cpp  from 2.0.12 to 2.0.14 seem to be tuned to support  a console for a  board that has a console associated with serial0.   I can change my code permanently to call serial0.end() before  serial0.begin is called if that is recommended as  a work around solution going forward.  If no code changes are planned, please suggest a way to allow use of serial0 with  user configurable  rx and tx pins  on boards like the LOLIN s2 mini that is unlikely to  break in future releases.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-19:
Issue with 2.0.14 confirmed. I'll test it with the code in the `release/v2.x` and, if necessary, fix it there for the 2.0.15.
This issue doesn't happen with 3.0.0-Alpha-3, which uses a refactored UART code.

@xalier1 - please stick to the 2.0.13 for now. 2.0.15 shall fix it and some other `HardwareSerial` issues.

xalier1, 2024-01-19:
@SuGlider - thanks for investigating this.  Just to make sure that you are seeing the same problem that I am, the problem also exists on 2.0.13. I had to revert to 2.0.12 for the problem to not occur.

SuGlider, 2024-01-19:
@xalier1 - yes. I'll dig into it and fix it for 2.0.15.

SuGlider, 2024-03-21:
Fix by https://github.com/espressif/arduino-esp32/pull/9373
To be released within 2.0.15

closed 2024-03-21 by SuGlider