espressif/arduino-esp32 #9120: (IDF Websocket client) - TRANSPORT_WS: Sec-WebSocket-Accept not found

zekageri, 2024-01-17:
### Board

ESP32-Wrover

### Device Description

A custom device

### Hardware Configuration

A custom hardware

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

The problem I'm facing is that i can't connect to my node.js server which is in the cloud with native websocket client.
It works my server is running on local, but i can connect to the server with Postman regardless if it is in the cloud or not.
I'm hosting my server at onrender.com.

Server is fine because it works if i connect to it with postman or frontend js

### Sketch

```cpp
#include <Arduino.h>
#include <esp_websocket_client.h>

void initWs(){
    esp_websocket_client_handle_t client;
    esp_websocket_client_config_t config = {};

    config.host = "ws://bridgetest-9q8c.onrender.com";
    config.transport = WEBSOCKET_TRANSPORT_OVER_TCP;

    client = esp_websocket_client_init(&config);
    esp_websocket_register_events(
        client,
        WEBSOCKET_EVENT_ANY,
        reinterpret_cast<esp_event_handler_t>(eventHandler),
        static_cast<void*>(this)
    );
    esp_websocket_client_start(client);
}
```

### Debug Message

```plain
E (260241) TRANSPORT_WS: Sec-WebSocket-Accept not found
E (260241) WEBSOCKET_CLIENT: Error transport connect
```

### Other Steps to Reproduce

As i said, i have tried with Postman and with various ws client simulators and it works on every way except my ESP32.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zekageri, 2024-01-17:
The server is https but it has an http interface and it does not redirect.
It is reachable with both ws and wss with postman and also with http and https

me-no-dev, 2024-01-17:
you should ask in the IDF repo. If you are on master here, then you are on IDF v5.1.2

zekageri, 2024-01-17:
Yeah, i figured i should ask it there, so i created an issue.
Strange things happening.

zekageri, 2024-01-18:
Can i somehow define `CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY` and `CONFIG_ESP_TLS_INSECURE` using platformIO?
I have tried to put these into `platformio.ini` like this

```
-D CONFIG_ESP_TLS_INSECURE
-D CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY
```

```
-D CONFIG_ESP_TLS_INSECURE=1
-D CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=1
```

```
-DCONFIG_ESP_TLS_INSECURE
-DCONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY
```

```
-DCONFIG_ESP_TLS_INSECURE=1
-DCONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=1
```

No success. The websocket client should work if i could define these.

closed 2025-01-15 by Parsaabasi