espressif/arduino-esp32 #9126: LittleFS upload "Success" but file read fails.

meakashrao, 2024-01-17:
### Board

ESP32-WROOM-32E-N4 

### Device Description

The hardware is custom, but I was able to update to and fro multiple Firmware Versions with the LITTLEFS Partitions.

### Hardware Configuration

NA

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

460800

### Description

1) Platform IO Serial Update
a) When I use Platform IO Core 6.1.13 and Home 3.4.4 with Espressif 32 v6.4.0, and upload firmware.bin file using the upload button and littlefs.bin using the upload FileSystem Image, my ESP is able to read the LITTLEFS file. 
b) When I upload using the OTA method by making the ESP go into Access Point Mode and use the files espota.py and esptool.py although both files are updated successfully my LittleFS read fails(mount doesn't but file read fails)
I tried to format the LITTLEFS partion using the function "LittleFS.format()" and it returns successful and uploaded the LITTLEFS.bin again and it still resulted in File not present and File read failed. 
c) Then if I take the ESP, erase the flash contents, and repeat step 'a' things work again. 

I am attaching a zip file with my files. 
[ESP_OTA_LITTLEFS_Details.zip](https://github.com/espressif/arduino-esp32/files/13964313/ESP_OTA_LITTLEFS_Details.zip)

I have multiple sections where I send the Error Message using BLE a
The Error Messages are 
a)  it says _"LITLFS-FileReadFail"_ in the function 
void readFile(fs::FS &fs, const char * path), which basically just reads and serial prints contents of the 2 files in the data folder in Platform IO
b) Files not found
c) Reading fails again. 

OTA upload Screenshots 
Firmware File 
[image]

SPIFFS Image:
[image]

Starting of the espota.py file
[image]

The whole espota.py and esptool.py are there in the zip

### Sketch

```cpp
I am attaching the LITTLEFS_Config.cpp here (it's included in the ZIP above as well)
[... 239 lines cut ...]
    Serial.print(array[i]);
    if (i < arraySize - 1) 
    {
      Serial.print(", ");
    }
  }
  Serial.println("]");
}

bool formatLittleFS()
{
    bool ret = LittleFS.format();
    return ret;
}
```

### Debug Message

```plain
NA.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

meakashrao, 2024-01-17:
Please help to decode the issue. Why does it works when I upload serially and why not with OTA 

My flatformio.ini file 
[image]

Jason2866, 2024-01-17:
Since there are changes in LittleFS version it is mandatory to know which core version you use. You don't use latest master, that is sure. Since no version is pinned in Platformio setup we don't know at all. So please change in your Platformio.ini  `platform = espressif32` to `platform = espressif32 @ 6.5.0` to use released core 2.0.14.

Do you still have the same behaviour with this setup?

meakashrao, 2024-01-17:
> Since there are changes in LittleFS version it is mandatory to know which core version you use. You don't use latest master, that is sure. Since no version is pinned in Platformio setup we don't [...]

I am using 6.4.0, I tried using 6.5.0 but that has some issue with the WiFi and Serial Port, so I am forced to stick to 6.4.0
This is what I get while compiling the properly working code on 6.5.0
[image]

meakashrao, 2024-01-17:
> Are you using in both cases the fs.img generated from Platformio? If yes, either the upload fails "somehow" -> the fs.img is loaded at a wrong address and/or the content got corrupted. To know what happened download the complete flash content with esptool.py First the working serial version. Upload via OTA. Dump this version. Compare both versions with a hex viewer to see where the differences are.

Thank you for this suggestion.
I found the below article to read the flash contents. 
[image]

Do you have any suggestions for what hex viewer to use? Also, this gives me .bin file, am I on the right track with this?

meakashrao, 2024-01-17:
> Are you using in both cases the fs.img generated from Platformio? If yes, either the upload fails "somehow" -> the fs.img is loaded at a wrong address and/or the content got corrupted. To know what [...]

Yes, I am using the same littlefs.bin file for Serial and OTA update.

I found this command to read the contents of the flash in a .bin extension
https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#read-flash-contents-read-flash

Am i right here? and do you have any recommendations for hex viewer?

Thanks for replying to my queries

meakashrao, 2024-01-17:
Also, is there any way to correct set the offset according to my min_spiffs.csv partition table during OTA using the command 
python .\espota.py -d -i 192.168.1.1 -p 3232 --auth=admin -f .\littlefs.bin -s

Jason2866, 2024-01-17:
Yes, correct for reading flash. Before changed to Mac used this one with Windows http://www.winhex.com/winhex/index-d.html

meakashrao, 2024-01-18:
Hello @Jason2866 

Should I be using "extra_scripts = ./littlefsbuilder.py" and the appropriate file as in this image 
[image]

or the line board_build.filesystem = littlefs in my .ini file does the job?

meakashrao, 2024-01-18:
> Are you using in both cases the fs.img generated from Platformio? If yes, either the upload fails "somehow" -> the fs.img is loaded at a wrong address and/or the content got corrupted. To know what [...]

So, I checked the bin files
First I took a machine on Which SPIFFS was never used then did OTA and read the flash.bin files 
Then uploaded it with serial and read the flash.bin files
Both have been written to the same location

With OTA there were multiple troubles
a) after updating the firmware.bin
[image]
This got solved after a Power ON OFF()spiffs mounted successfully)

b) after updating littefs.bin
[image]

but the same machine after serial update works properly.

meakashrao, 2024-01-18:
As suggested I have check the flash contents 

I have attached a zip folder which contains 
1) ota_flash-machine2_dump.bin -> Bin file read from flash after OTA update of both firmware and littlefs files
2) serial_flash-machine2_dump.bin -> Bin file read from flash after serial flashing of both firmware and littlefs files
3) Serial and OTA update.docx -> this contains all screenshots of Serial terminal and the screenshots of above bin files
4) firmware.bin -> This is the actual firmware.bin being used for OTA and serial update
5) littlefs.bin -> This is the actual firmware.bin being used for OTA and serial update
6) LITTLEFS_Config.cpp -> This is my LittleFS config file
7) platformio_ini -> PlatformIO configuration File
8) espota.py -> file that I am using
9) esptool.py -> file that I am using 

Please if someone can find time to help me with what the issue is with my code.  

[LittleFs_Debug.zip](https://github.com/espressif/arduino-esp32/files/13974216/LittleFs_Debug.zip)

Jason2866, 2024-01-18:
The entry "extra_scripts = ./littlefsbuilder.py" is not needed anymore with actual espressif32 Platformio Platform. This part selects "mklittlefs" to use as App to build the fs img.
https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L264-L281
This part is working correctly since flashing via VSC results in working firmware and accessible files from Filesystem.
So "something" bad happens via the OTA update of the FS. When doing this the flash content of the used flash differs from the one via wired flash. Can't help here any further since not using `espota.py`

meakashrao, 2024-01-18:
> The entry "extra_scripts = ./littlefsbuilder.py" is not needed anymore with actual espressif32 Platformio Platform. This part selects "mklittlefs" to use as App to build the fs img. [...]

Thank you so much for the insights. I hope someone else picks up and helps me

Jason2866, 2024-01-18:
I think it is not an Platformio issue. The used `espota.py` is the one provided from the framework in folder "tools". 
@me-no-dev Has it been tested lately if uploading a LittleFS image via `espota.py` is working? Flashing the same LittleFS image via esptool.py (wired) does work.

meakashrao, 2024-01-22:
Hello Team, any update on this?

closed 2024-01-22 by meakashrao

Jason2866, 2024-01-22:
@meakashrao Great you find the reason. "Nice" trap.