espressif/arduino-esp32 #9136: Incorrent timezone in `struct tm` for `gmtime_r`

qistoph, 2024-01-18:
### Board

ESP32 Dev Module

### Device Description

Plain module on board

### Hardware Configuration

N/A

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Ubuntu

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

A `struct tm` contains incorrect timezone information when filled by `gmtime_r`.

Expected output of Sketch below:
```
GMT: Thursday, January 18 2024 15:52:52 Zone: GMT (+0000)
Local: Thursday, January 18 2024 16:52:52 Zone: CET (+0100)
```

Actual output:
```
GMT: Thursday, January 18 2024 15:52:52 Zone: CET (+0100)
Local: Thursday, January 18 2024 16:52:52 Zone: CET (+0100)
```

### Sketch

```cpp
#include <WiFi.h>
#include <time.h>
#include "WiFiCreds.h"

const char* ssid = WIFI_SSID;
const char* wifipw = WIFI_PASS;

#define TZ_Europe_Amsterdam PSTR("CET-1CEST,M3.5.0,M10.5.0/3")

void  startWifi(){
  WiFi.begin(ssid, wifipw);
  Serial.println("Connecting Wifi");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.print("Wifi RSSI=");
  Serial.println(WiFi.RSSI());
}

void setTimezone(String timezone){
  Serial.printf("  Setting Timezone to %s\n",timezone.c_str());
  setenv("TZ",timezone.c_str(),1);  //  Now adjust the TZ.  Clock settings are adjusted to show the new local time
  tzset();
}

void setupTime(String timezone){
  struct tm timeinfo;

  Serial.println("Setting up time");
  configTime(0, 0, "pool.ntp.org");    // First connect to NTP server, with 0 TZ offset
  if(!getLocalTime(&timeinfo)){
    Serial.println("  Failed to obtain time");
    return;
  }
  Serial.println("  Got the time from NTP");
  setTimezone(timezone);
}

void printLocalTime(){
  time_t now;
  struct tm timeinfo;

  time(&now);
  Serial.printf("Now: %ld\n", now);

  gmtime_r(&now, &timeinfo);
  Serial.println(&timeinfo, "GMT: %A, %B %d %Y %H:%M:%S Zone: %Z (%z)");

  localtime_r(&now, &timeinfo);
  Serial.println(&timeinfo, "Local: %A, %B %d %Y %H:%M:%S Zone: %Z (%z)");
}

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  startWifi();

  setupTime(TZ_Europe_Amsterdam);   // Set for Amsterdam/NL
  printLocalTime();
}

void loop() {
  printLocalTime();
  delay(1000);
}
```
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-01-18:
tm doesn't include timezone info, so print() is still using the set locale.  You can see what Print is doing at https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Print.cpp#L179.  Use [strftime_l](https://man7.org/linux/man-pages/man3/strftime.3.html) if you want to give it a locale parameter.

VojtechBartoska, 2024-05-23:
Closing as expired.

closed 2024-05-23 by VojtechBartoska