espressif/arduino-esp32 #9137: rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) when connecting NodeMCU-32S module to NRF2401 module.

NibaliDev, 2024-01-18:
### Board

NodeMCU-32S 

### Device Description

The NodeMCU-32S module is directly connectd to the NRF2401 module via jumper cables. 

### Hardware Configuration

CE: 12 
CSN: 23 
SCK: 15 
MISO: 14 
MOSI: 2 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 

### Operating System

Windows 10 

### Flash frequency

89 MHz

### PSRAM enabled

yes

### Upload speed

115200 

### Description

When I try to communicate with an NRF2401 radio module, theough SPI, with an NodeMCU-32S development board, I get the following error. The watchdog time treiggerad a system reset after something unexpected happend. I only get this error when  the SPI connection is plugged into the NRF2401 module, even if then module is not powered on. The error also says SPI_FAST_FLASH_BOOT, before thw board gets reseted again. This error is contiously printed out, efetr every reset. Beacause of this, I think that it must have something to do with the SPI connection. The board maybe can't access the flash memory because it's ISP ports is aleeary used or somtehing. I'm prett sure thats the prblem, but I have too little experience and knowledge about the ESP32 to troubleshott all this by myself. I would be  ery thankful idf someone cvould help me solve this. 

### Sketch

```cpp
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(4, 2); // CE, CSN

const byte address[6] = "00001"; 

void setup() {
  Serial.begin(115200); 
  radio.begin();
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_MIN);
  radio.startListening();
}

void loop() {
  if (radio.available()) {
    char text[32] = "";
    radio.read(&text, sizeof(text));
    Serial.println(text); 
  }
}
```

### Debug Message

```plain
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 988
ets_main.c 384 
ets Jul 29 2019 12:21:46
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-01-18:
The library is blocking the cpu, so the watchdog reboots the system. Look for another library that is known to support esp32.

NibaliDev, 2024-01-22:
Thank for responding so fast. I'm very sorry that I took so long time responding, but I have now found the problem. It has nothing to do with the library, I realized that after repeating the exact same error, when no code was uploaded to the ESP32 mcirocontroller. The problem only appeared when the ESP32 was connected to the NRF24. After some troubleshooting, I found out that the problem acutally was very silly. I had read the text wrong and accidentily connected GND to CMD on the NodeMCU development board. I didn't se that, and obviosuly thought that the power conenctions would be my least problem. It is very obvious that the watchdog triggers a reset when this is happening, and I'm actually kind of sorry for even publishing this error and took your time. Thanks for your commitment.

closed 2024-01-22 by NibaliDev

samrepublicc, 2024-08-21:
i managed to get esp32 cam with doing somethin. Guys i spent hours and i found that you should connect power suply with 5v not 3.3v and make sure that you connected 5v to 5v when you are using ftdi module.

palmarci, 2024-09-02:
> Thank for responding so fast. I'm very sorry that I took so long time responding, but I have now found the problem. It has nothing to do with the library, I realized that after repeating the exact [...]

Oh my god, thank you! I though I was going insane, I even borrowed a logic analyser from work, but I have never checked for continuity because I read "GND".  The text is so blurry.