espressif/arduino-esp32 #9140: WIFI_IF_AP, WIFI_PROTOCOL_LR / UDP not working in 2.0.14 but 2.0.3

sartyx, 2024-01-19:
### Board

ESP32-S2 / ESP32 DEV Module

### Device Description

Lolin S2 Mini and Sunton ESP32-2432S024

### Hardware Configuration

Nothing for the S2-Mini but the display got many embedded parts, not sure if this matters as the examples do work in 2.0.3.

### Version

v2.0.14

### IDE Name

Arduino IDE, Platformio, same results on both IDE

### Operating System

Windows 10

### Flash frequency

no idea

### PSRAM enabled

yes

### Upload speed

115200

### Description

I tried this code to set up a communication between the two ESP32.
It works quite well using ESP 2.0.3 but fails on 2.0.14.
Both sides master/slave compile but the client fails to connect usind 2.0.14

### Sketch

```cpp
https://github.com/jnogues/ESP32-Long-Range-WiFi

/////////////////////////////////////////////////////////////////////////////////
// MASTER
// based on https://gist.github.com/yaqwsx/ac662c9b600ef39a802da0be1b25d32d
// 2018.07.14 jnogues@gmail.com, Jaume Nogués, rPrim Tech
// This sketch shows the use of 802.11 LR (Low Rate)
// master.ino

#include <Arduino.h>
#include <WiFi.h>
#include <WiFiUdp.h>
#include <esp_wifi.h>

const char* ssid = "kkkkk";//AP ssid
const char* password = "12345678";//AP password
const char* ssidRouter = "MySSID";//STA router ssid
const char* passwordRouter = "MyPASSWORD";//STA router password
WiFiUDP udp;

void setup() {
    pinMode(5, OUTPUT);//builtin Led, for debug
    digitalWrite(5, HIGH);
    Serial.begin( 115200 );
    Serial.println( "Master" );

    //first, we start STA mode and connect to router
    WiFi.mode( WIFI_AP_STA );
    WiFi.begin(ssidRouter,passwordRouter); 
    
    //Wifi connection
    while (WiFi.status() != WL_CONNECTED) 
    {
      delay(500);
      Serial.print(".");
    }

    Serial.println("Router WiFi connected");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  
    //second, we start AP mode with LR protocol
    //This AP ssid is not visible whith our regular devices
    WiFi.mode( WIFI_AP );//for AP mode
    //here config LR mode
    int a= esp_wifi_set_protocol( WIFI_IF_AP, WIFI_PROTOCOL_LR );
    Serial.println(a);
    WiFi.softAP(ssid, password);
    Serial.println( WiFi.softAPIP() );
    Serial.println("#");//for debug
    delay( 1000 );
    digitalWrite(5, LOW); 
    udp.begin( 8888 );
}

void loop() 
{
    udp.beginPacket( { 192, 168, 4, 255 }, 8888 );//send a broadcast message
    udp.write( 'b' );//the payload
    digitalWrite(5, !digitalRead(5));
    
    if ( !udp.endPacket() ){
        Serial.println("NOT SEND!");
        delay(100);
        ESP.restart(); // When the connection is bad, the TCP stack refuses to work
    }
    else{
          Serial.println("SEND IT!!"); 
    }     
    
    delay( 1000 );//wait a second for the next message
}

///////////////////////////////////////////////////////////////////////////////
// SLAVE
// based on https://gist.github.com/yaqwsx/ac662c9b600ef39a802da0be1b25d32d
// 2018.07.14 jnogues@gmail.com, Jaume Nogués, rPrim Tech
// This sketch shows the use of 802.11 LR (Low Rate)
// slave.ino

#include <Arduino.h>
#include <WiFi.h>
#include <WiFiUdp.h>
#include <esp_wifi.h>

const char* ssid = "kkkkk";//AP ssid
const char* password = "12345678";//AP password

WiFiUDP udp;

const char *toStr( wl_status_t status ) {
    switch( status ) {
    case WL_NO_SHIELD: return "No shield";
    case WL_IDLE_STATUS: return "Idle status";
    case WL_NO_SSID_AVAIL: return "No SSID avail";
    case WL_SCAN_COMPLETED: return "Scan compleded";
    case WL_CONNECTED: return "Connected";
    case WL_CONNECT_FAILED: return "Failed";
    case WL_CONNECTION_LOST: return "Connection lost";
    case WL_DISCONNECTED: return "Disconnected";
    }
    return "Unknown";
}

void setup() {
    Serial.begin( 115200 );
    Serial.println( "Slave" );
    pinMode(5, OUTPUT);//bultin Led, for debug

    //We start STA mode with LR protocol
    //This ssid is not visible whith our regular devices
    WiFi.mode( WIFI_STA );//for STA mode
    //if mode LR config OK
    int a= esp_wifi_set_protocol( WIFI_IF_STA, WIFI_PROTOCOL_LR );
    if (a==0)
    {
      Serial.println(" ");
      Serial.print("Error = ");
      Serial.print(a);
      Serial.println(" , Mode LR OK!");
    }
    else//if some error in LR config
    {
      Serial.println(" ");
      Serial.print("Error = ");
      Serial.print(a);
      Serial.println(" , Error in Mode LR!");
    }
      
    WiFi.begin(ssid, password);//this ssid is not visible

    //Wifi connection, we connect to master
    while (WiFi.status() != WL_CONNECTED) 
    {
      delay(500);
      Serial.print(".");
    }

    Serial.println("WiFi connected");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  
    udp.begin( 8888 );
}

void loop() {
     //problems whith connection
    if ( WiFi.status() != WL_CONNECTED ) 
    {
        Serial.println( "|" );
        int tries = 0;
        WiFi.begin( ssid, password );
        while( WiFi.status() != WL_CONNECTED ) {
            tries++;
            if ( tries == 5 )
                return;
            Serial.println( toStr( WiFi.status() ) );
            delay( 1000 );
        }
        Serial.print( "Connected " );
        Serial.println( WiFi.localIP() );
    }
    //if connection OK, execute command 'b' from master
    int size = udp.parsePacket();
    if ( size == 0 )
        return;
    char c = udp.read();
    if ( c == 'b' ){
        digitalWrite(5, !digitalRead(5));//toggle Led
        Serial.println("RECEIVED!");
        Serial.println(millis());
    }
    udp.flush();
}
```

### Debug Message

```plain
no debug message, the client is just not connecting
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-01-19:
Hi @sartyx, can you please set Debug level to Verbose on both boards and post the Debug message? Thanks

sartyx, 2024-01-19:
[esp32log.txt](https://github.com/espressif/arduino-esp32/files/13988790/esp32log.txt)

sure, it is huge though... the dev board was flashed at 80MHz as I just noticed

closed 2025-01-15 by Parsaabasi