espressif/arduino-esp32 #9145: Enhancement of IPAddress to save some bytes by encoding IPv4 as IPv4-mapped addresses

sgryphon, 2024-01-19:
### Related area

Networking - IPAddress

### Hardware specification

No

### Is your feature request related to a problem?

No

### Describe the solution you'd like

The IPv4-mapped address range is specifically designed for storing IPv4 addresses in an IPv6 structure, i.e. they actually represent IPv4 addresses.

This means storing IPType as a separate field is not necessary. IPType::IPV4 is determined by starting with 80 zero bits and then having 0xFFFF as the next 16 bits.

These improvements should also be pushed up to the Arduino Core project.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

sgryphon, 2024-02-20:
Note one drawback of this approach is that the default (zeroed) value is :: (IPv6 all zeros), and no longer ::ffff:0000 aka 0.0.0.0, which it was previously.

me-no-dev, 2024-04-03:
I can agree as long as upstream Arduino API also implements this

closed 2025-01-15 by Parsaabasi