espressif/arduino-esp32 #9148: Don't know where the 10ms delay comes from? 【不知道哪里来的10ms延时？】

mysteriousTreasure, 2024-01-20:
### Board

ESP32-S3  N16R8

### Device Description

Using the software environment,【使用软件环境，】
vscode+platformIO arduino
espressif32 6.4.0
Use hardware【硬件使用】：ESP32-S3 N16R8

Question:
There is an inexplicable 10ms do not know what task is being performed.
【问题：
出现莫名其妙的10ms不知道在执行什么任务。】

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

windows11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Question:
There is an inexplicable 10ms do not know what task is being performed.
【问题：
出现莫名其妙的10ms不知道在执行什么任务。】

for example
【举个简单例子】

`#include <Arduino.h>
unsigned long lasttime = 0;
unsigned long nowtime = 0;
void setup()
{
Serial.begin(115200);
}

void loop()
{
lasttime = micros();
Serial.printf("1->lasttime=%u,", lasttime);

nowtime = micros() - lasttime;
Serial.printf("%u\n", nowtime);

vTaskDelay(1);
}`

[image]

And if vTaskDelay(1); Change to vTaskDelay(2). There is no such phenomenon
The output is normal.
【而如果vTaskDelay(1);修改为vTaskDelay(2);就没这个现象
输出很正常】
[image]

There is also a normal writing output, or vTaskDelay(1);
However, when you do not use a string of more than two bytes to print, it is normal
【还有一种写法也输出正常，还是vTaskDelay(1);但是不使用2个字节以上的字符串打印时候，也正常】
[image]

And when I thought I had found the problem, what was this routine?
【而我满心欢喜以为找到问题时候，结果这个例程又是为什么呢？】
[image]

### Sketch

```cpp
#include <Arduino.h>
unsigned long lasttime = 0;
unsigned long nowtime = 0;
void setup()
{
Serial.begin(115200);
}

void loop()
{
lasttime = micros();
Serial.printf("1->lasttime=%u,", lasttime);

nowtime = micros() - lasttime;
Serial.printf("%u\n", nowtime);

vTaskDelay(1);
}
```

### Debug Message

```plain
Question:
There is an inexplicable 10ms do not know what task is being performed.
【问题：
出现莫名其妙的10ms不知道在执行什么任务。】

for example
【举个简单例子】

`#include <Arduino.h>
unsigned long lasttime = 0;
unsigned long nowtime = 0;
void setup()
{
Serial.begin(115200);
}

void loop()
{
lasttime = micros();
Serial.printf("1->lasttime=%u,", lasttime);

nowtime = micros() - lasttime;
Serial.printf("%u\n", nowtime);

vTaskDelay(1);
}`

![1](https://github.com/espressif/arduino-esp32/assets/95865062/6e647beb-acf7-4420-8796-e40b4e4445a5)

And if vTaskDelay(1); Change to vTaskDelay(2). There is no such phenomenon
The output is normal.
【而如果vTaskDelay(1);修改为vTaskDelay(2);就没这个现象
输出很正常】
![2](https://github.com/espressif/arduino-esp32/assets/95865062/642f23e2-bade-4793-b79a-65f47e49b480)

There is also a normal writing output, or vTaskDelay(1);
However, when you do not use a string of more than two bytes to print, it is normal
【还有一种写法也输出正常，还是vTaskDelay(1);但是不使用2个字节以上的字符串打印时候，也正常】
![3](https://github.com/espressif/arduino-esp32/assets/95865062/9cb57763-14b5-43fe-b490-02008b651e95)

And when I thought I had found the problem, what was this routine?
【而我满心欢喜以为找到问题时候，结果这个例程又是为什么呢？】
![4](https://github.com/espressif/arduino-esp32/assets/95865062/5142eec8-74a1-4f51-a710-ae43921133ab)
```

### Other Steps to Reproduce

..

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-01-20:
This is due to how delay is calculated.  The minimum delay is a multiple of the FreeRTOS tick frequency (1000 Hz).
https://github.com/espressif/arduino-esp32/issues/2981

ChiayuK, 2024-02-16:
Is this problem has been solved? I also meet this question

mysteriousTreasure, 2024-02-19:
> Is this problem has been solved? I also meet this question

No solution

TD-er, 2024-02-19:
> > Is this problem has been solved? I also meet this question
> 
> No solution

Are you sure?
As I mentioned 3 weeks ago, you are sending about 24 bytes via serial every msec.

At 115200 baud, you can't process this amount of data and thus at some point the serial buffer will be full and the serial print statement will be blocking.
As simple 'proof' of this theory, you can try to only send the last few bytes to serial, or increase the baud rate.

VojtechBartoska, 2024-05-23:
Hello, closing as therr are no action points and this issue is just a question.

closed 2024-05-23 by VojtechBartoska