espressif/arduino-esp32 #9149: setting CPU Frequency in Arduino IDE

sboicu, 2024-01-21:
### Board

ESP32

### Device Description

n.a.

### Hardware Configuration

n.a.

### Version

v2.0.14

### IDE Name

ARDUINo IDE

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I try to build a sketch with CPU Freq of 80Mhz

[image]

But if I check the sdk config file generated I see this setting:

[image]

Does this mean that the project is build with the 160Mhz? or maybe is there another location where a new sdk for a specific project is generated?

thanks

### Sketch

```cpp
n.a.
```

### Debug Message

```plain
FQBN: esp32:esp32:esp32:CPUFreq=80,PartitionScheme=min_spiffs
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-01-21:
The sdkconfig file shows the settings which where used when compiling the Arduino code with IDF. It has nothing to do with the settings you choose in ArduinoIDE.

sboicu, 2024-01-21:
> The sdkconfig file shows the settings which where used when compiling the Arduino code with IDF. It has nothing to do with the settings you choose in ArduinoIDE.

do you know if Arduino generates a sdkconfig file for the projects?

Jason2866, 2024-01-21:
There is no sdkconfig file generated when using Arduino

lbernstone, 2024-01-21:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-cpu.h#L40

closed 2024-05-23 by VojtechBartoska