espressif/arduino-esp32 #9152: ESP32 unable to connect to WiFi AP (Raspberry PI with RTL8812AU WiFi module) but connects to "classic" WLAN Router

nocs85, 2024-01-22:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

DOIT ESP32 DEVKIT V1

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.2.1

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hello, I am trying a very basic WiFi connection example with Arduino IDE (Examples -> WiFi -> WiFiClientConnect).
I only changed "ssid+password", rest of the Example code is unchanged.
I am trying to connect to an AP that is a Raspberry Pi 4, with Raspbian "Bullseye" and a USB WiFi dongle "TP-Link Archer T2U PLUS [RTL8821AU]" configured as AP.
The ESP32 is not able to connect to the "Raspberry Pi" AP.
For the sketch output please see "Debug Message" section.

The strange thing is:
* My Windows 11 PC can successfully connect to the Raspberry Pi AP
* The same ESP32 with the same sketch (just changed ssid+password) is able to connect to my "classic" WLAN Router

This leads me to believe there is some handshaking with the Raspberry Pi AP that ESP32 is not able to handle correctly.

I have tried Arduino ESP32 v2.0.14 and also 3.0.0 Alpha 3, both have the same problem connecting to the Raspberry Pi AP.

I am open to providing any necessary information to help diagnose and solve this problem.
Thank you!

### Sketch

```cpp
/* Wi-Fi STA Connect and Disconnect Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
   
*/
#include <WiFi.h>

const char* ssid     = "************";
const char* password = "************";

int btnGPIO = 0;
int btnState = false;

void setup()
{
    Serial.begin(115200);
    delay(10);

    // Set GPIO0 Boot button as input
    pinMode(btnGPIO, INPUT);

    // We start by connecting to a WiFi network
    // To debug, please enable Core Debug Level to Verbose

    Serial.println();
    Serial.print("[WiFi] Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);
// Auto reconnect is set true as default
// To set auto connect off, use the following function
//    WiFi.setAutoReconnect(false);

    // Will try for about 10 seconds (20x 500ms)
    int tryDelay = 500;
    int numberOfTries = 20;

    // Wait for the WiFi event
    while (true) {
        
        switch(WiFi.status()) {
          case WL_NO_SSID_AVAIL:
            Serial.println("[WiFi] SSID not found");
            break;
          case WL_CONNECT_FAILED:
            Serial.print("[WiFi] Failed - WiFi not connected! Reason: ");
            return;
            break;
          case WL_CONNECTION_LOST:
            Serial.println("[WiFi] Connection was lost");
            break;
          case WL_SCAN_COMPLETED:
            Serial.println("[WiFi] Scan is completed");
            break;
          case WL_DISCONNECTED:
            Serial.println("[WiFi] WiFi is disconnected");
            break;
          case WL_CONNECTED:
            Serial.println("[WiFi] WiFi is connected!");
            Serial.print("[WiFi] IP address: ");
            Serial.println(WiFi.localIP());
            return;
            break;
          default:
            Serial.print("[WiFi] WiFi Status: ");
            Serial.println(WiFi.status());
            break;
        }
        delay(tryDelay);
        
        if(numberOfTries <= 0){
          Serial.print("[WiFi] Failed to connect to WiFi!");
          // Use disconnect function to force stop trying to connect
          WiFi.disconnect();
          return;
        } else {
          numberOfTries--;
        }
    }
}

void loop()
{
    // Read the button state
    btnState = digitalRead(btnGPIO);
    
    if (btnState == LOW) {
      // Disconnect from WiFi
      Serial.println("[WiFi] Disconnecting from WiFi!");
      // This function will disconnect and turn off the WiFi (NVS WiFi data is kept)
      if(WiFi.disconnect(true, false)){
        Serial.println("[WiFi] Disconnected from WiFi!");
      }
      delay(1000);
    }
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[    31][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

[WiFi] Connecting to ***********
[    78][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   145][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] Failed to connect to WiFi![ 10654][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 10654][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 203 - ASSOC_FAIL
[ 10659][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
```

### Other Steps to Reproduce

I have tried Arduino ESP32 v2.0.14 and also 3.0.0 Alpha 3, both have the same problem connecting to the Raspberry Pi AP.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

nocs85, 2024-01-22:
Unfortunately I can confirm that the AP network is set to 2.4Ghz:
This is the output of "iwconfig" from raspberry pi command line:

```
wlan1     IEEE 802.11bg  ESSID:"********"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:2.437 GHz  Access Point: 6C:5A:B0:D0:4A:87
          Bit Rate:54 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

And also confirmation from Windows 11 WiFi properties for that network:
[image]

nocs85, 2024-01-24:
Hi, so I just wanted to see if a ESP8266 can connect to the AP. It's a WEMOS D1 mini.
I know it's most probably a different base-system regarding the libraries interfacing the ESP but I thought I'll just give it a try.
I run this simple code:

```
#include <ESP8266WiFi.h>        // Include the Wi-Fi library

const char* ssid     = "*******";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "******";     // The password of the Wi-Fi network

void setup() {
  Serial.begin(115200);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');
  
  WiFi.begin(ssid, password);             // Connect to the network
  Serial.print("Connecting to WiFi AP ...");

  int i = 0;
  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(1000);
    Serial.print(++i); Serial.print(' ');
  }

  Serial.println('\n');
  Serial.println("Connection established!");  
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());         // Send the IP address of the ESP8266 to the computer
}

void loop() { }
```

And it connected successfully to the AP and got an IP address.
This is the output on the serial monitor:

```
Connection established!
IP address:	10.0.1.23
```

nocs85, 2024-01-24:
I did also the following test:

1. Started the ESP32 WiFiClientConnect sketch.
2. The ESP32 could not connect.
3. Restarted the Raspberry Pi and after Raspbian boot-up the the ESP32 plotted the following:

```
 80003][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 80006][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
[ 82428][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 82428][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 82431][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
[ 83497][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 87038][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 87039][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 10.0.1.22, MASK: 255.255.255.0, GW: 10.0.1.1
```

If I subsequently push the "EN"-button on the ESP32-DevKit board, the ESP32 again is NOT able to connect to the Raspberry Pi AP with the same Serial Monitor log as in the first post.

I did multiple time the "Raspberry Pi AP" reboot-test and the ESP32 behavior is reproducible.
Each time after a "Raspberry Pi AP" reboot, the ESP32 automatically connects once to the AP. On every subsequent EN-button push it can't connect to the AP.

lbernstone, 2024-01-25:
With that sketch, if you hold the boot (prg) button down after it fails, you should see it say that it is disconnecting from wifi.  That should make it clear any saved lease. I would also restart the hostapd/dnsmasq and clear the lease file (maybe /var/lib/misc/dnsmasq.leases).  Post the verbose log from the first connection there if it fails.

nocs85, 2024-01-25:
After a failed connection, if I press the boot-button, the following output is printed on the serial monitor:

```
[WiFi] Disconnecting from WiFi!
[ 39608][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 39608][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 203 - ASSOC_FAIL
[ 39613][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
[ 39620][E][WiFiGeneric.cpp:1272] mode(): Could not set mode! 12308
[ 39626][E][WiFiSTA.cpp:299] begin(): STA enable failed!
[ 39631][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
[WiFi] Disconnected from WiFi!
```

If i restart hostapd via "sudo systemctl restart hostapd" then the ESP32 is able to connect once to the WiFi AP. Further connections to the AP are not possible:
```
[ 19067][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 19067][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 203 - ASSOC_FAIL
[ 19071][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
[ 21493][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 21493][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 21497][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
[ 22547][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 26086][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 26086][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 10.0.1.22, MASK: 255.255.255.0, GW: 10.0.1.1
```

me-no-dev, 2024-01-25:
if restarting hostapd is allowing the ESP to connect, then I guess it's something on the PI side that is preventing the ESP to connect. It's trying... but is failing to assoc with the AP. Any chance you can get some logs off of the Pi? Can you also share your AP configuration (if any)?

nocs85, 2024-01-25:
I can provide whatever is needed.
I can share the hostapd.conf if that is helpful or any other file, please just mention what you need for your analysis.
I can get logs of the Pi, just say which one you need.

nocs85, 2024-01-25:
Please bear in mind that the ESP8266 with a simple sketch can connect multiple times without problems

me-no-dev, 2024-01-26:
it's best to provide everything necessary to replicate the issue and all logs that are in any way connected to the problem

nocs85, 2024-01-26:
Since I am no expert regarding WiFi on ESP32 and on Raspberry Pi I am not really sure which logs would be useful for the analysis. If you could mention which logs (on Raspberry Pi) are necessary I'm more than happy to provide them.

lbernstone, 2024-01-26:
Does it consistently work if you put `WiFi.disconnect(false,true);` before the begin?

nocs85, 2024-01-26:
This code doesn't change the connection behaviour of the sketch at all.
The previous observed problems all remain.
```
WiFi.disconnect(false,true);
WiFi.begin(ssid, password);
```

nocs85, 2024-01-26:
I have enabled verbose logging of hostapd and dnsmasq on the raspberry pi.
hostapd doesn't print anything, not on successful connection and not on unsuccessful connection. It just says that it successfully started the AP.
dnsmasq only prints something on successful connection:

```
Jan 26 18:09:37 dnsmasq-dhcp[1204]: DHCP packet received on wlan1 which has no address
Jan 26 18:09:38 dnsmasq-dhcp[1204]: DHCP packet received on wlan1 which has no address
Jan 26 18:09:39 dnsmasq-dhcp[1204]: DHCPDISCOVER(wlan1) 24:6f:28:17:3f:44
Jan 26 18:09:39 dnsmasq-dhcp[1204]: DHCPOFFER(wlan1) 10.0.1.22 24:6f:28:17:3f:44
Jan 26 18:09:39 dnsmasq-dhcp[1204]: DHCPREQUEST(wlan1) 10.0.1.22 24:6f:28:17:3f:44
Jan 26 18:09:39 dnsmasq-dhcp[1204]: DHCPACK(wlan1) 10.0.1.22 24:6f:28:17:3f:44 esp32-173F44
```

nocs85, 2024-02-01:
Hello, can I provide some more information that is useful for further debugging?

lbernstone, 2024-02-01:
Turn on all the logging in /etc/config/dhcp
```
option logfacility '/tmp/log/dnsmasq.log'
option logdhcp '1'
option logqueries '1'
option quietdhcp '1'
```
Post the log of when it fails (matching the 203/201 errors as above).
Is the esp32 powered by the onboard ldo?  Do you have anything else drawing power from that? Do you have any other devices connected?

nocs85, 2024-02-02:
I couldn't find /etc/config/dhcp in Raspbian but I have added the following 2 options to  /etc/dnsmasq.conf 

```
log-queries
log-dhcp
```
Unfortunately, dnsmasq doesn't log anything when a connection attempt by ESP32 fails.
It only logs around 40 lines in the log-file when the connection is successful.

The ESP32 is powered via USB. No peripherals connected to the ESP32-DevKit Board, it's just the Board and nothing else.

I also tried with a static ip:

```
  IPAddress ip(10, 0, 1, 52);
  IPAddress gw(10, 0, 1, 1);    
  IPAddress sub(255, 255, 255, 0);    
  WiFi.config(ip, gw, sub);
  WiFi.begin(ssid, password);
```

And there are the same problems as with dynamic IP. First attempt is successful:
```
[ 25097][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 25099][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 25101][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 10.0.1.52, MASK: 255.255.255.0, GW: 10.0.1.1
```

Futher attempts are all without success:
```
[WiFi] Failed to connect to WiFi![ 10629][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 10629][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 203 - ASSOC_FAIL
[ 10634][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
```

me-no-dev, 2024-02-02:
you do not have an IP problem, so checking DHCP is pointless. It is failing to connect to the AP, so you need to check the wifi logs. I can not tell which logs you need, because i do not have such setup and you did not provide any pointers on how to configure one.

Calliska, 2024-02-09:
I faced with the same problem. There is no difference whether I used built-in Raspberry WiFi module or external - ESP32 didn`t connected to the Raspberry access point.
The solution that I found is to use Ubuntu not Raspbian.

nocs85, 2024-04-04:
I'd like to give a short update:
I tried with a WiFi dongle with rtl8812bu chipset and the problem is not present.
The problem evidently has to do something with the rtl8812au driver and with the interaction between ESP32 and the rtl8812au AP.

If there is interest I can provide informations to further analyze the problem or otherwise we can also close the Ticket.

simonkaemmer, 2024-09-05:
Had a simmilar issue, but I was using micropython on ESP32 and NetworkManager to manage AP on the RaspberryPi. After days of research, I tried to use Ubuntu Server (6.8.0-1010-raspi) instead of Raspberry Pi OS. Sadly the same issue occured. 

Somehow I messed around with **hostapd** and **dnsmasq** as an alternative for NetworkManager. I deactivated **systemd-resolved** to replace it with **dnsmasq** and finally the ESP was able to connect to my raspberry.

You can configure **dnsmasq** to act as the main DNS-Resolver quite easy:

Add the following line to '''/etc/dnsmasq.con''':
'''server=8.8.8.8'''.

simonkaemmer, 2024-09-05:
Had a similar issue with following setup:

Raspberry Pi 4b (Raspberry Pi OS "Bookworm") using NetworkManager to configure the AccessPoint.
Espressiv ESP32 WROOM using Micropython.

When trying to connect, the ESP logs "Wifi: Internal Error"

No matter what I tried, the ESP was never able to connect to the rapsberry AP.
After some research, I switched to Ubuntu Server(6.8.0-1010-raspi). 

As a last resort, I tried to replace the network-manager with **hostapd** and **dnsmasq**. A port-conflict between **dnsmasq** and **systemd-resolved** lead me to deactivate and mask **sytemd-resolved**. Somehow, the ESP was now able to connect. After reactivating **systemd-resolved** the old issue occured again. 

So I replaced **systemd-resolved** with **dnsmasq** and configured it to work as a dns-resolver. 

Finally everything worked as wanted.

physikant, 2024-11-26:
I'm having the same issue with a Raspberry Pi 4 and an ESP32. Seems like the ESP sees the access point as using WPA security only (i.e., no WPA2 or WPA3). I observed this when running a WiFi scan programm on the ESP.

Putting 
`WiFi.setMinSecurity(WIFI_AUTH_WPA_PSK); `

before 
`Wifi.begin();`

seems to help. 
However, I still don't know if this is a Raspi problem or an ESP problem...