espressif/arduino-esp32 #9154: Gpio pinMode error

Ivanbk, 2024-01-22:
### Board

ESP32 Wroom DevKit & Lolin32 Lite

### Device Description

Esp32 in breadboard with 3 multiplexers connected

### Hardware Configuration

//Read Mux 1 & 2
#define ReadS0 16
#define ReadS1 4
#define ReadS2 2 
#define ReadS3 15    
#define ReadEnable 17
#define ReadAnalogPin 35
#define ReadDigitalPin 34
//Out Mux 
#define OutS0 21
#define OutS1 19  
#define OutS2 18    
#define OutS3 5
#define OutEnable 22
#define OutPin 32

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Weird Behavior when setting consecutive pinMode as OUTPUT, i get random crashes and random error in pin setting and sometimes random Serial characters and freezing of sketch. For example, sometimes my code does not initialize when starting the set pin section, other times some pins give error, but the pins with the errors are not consistent, sometimes it's a set of pins, sometimes other pins. Completely randomly.

`E (4607) gpio: GPIO can only be used as input mode
[  4613][E][esp32-hal-gpio.c:130] __pinMode(): GPIO config failed`

### Sketch

```cpp
pinMode(mux_read_enable_, OUTPUT);
    pinMode(mux_out_enable_, OUTPUT);
    pinMode(mux_read_analog_, INPUT);
    pinMode(mux_read_digital_, INPUT);
    pinMode(mux_out_pin_, OUTPUT);
    Serial.println("Mux Pins Set");
    for (int i = 0; i < 4; i++) {
        Serial.println(mux_read_s[i]);
        pinMode(mux_read_s_[i], OUTPUT);
        //delay(10);
    }
    Serial.println("Mux Read Pins Set");
    for (int i = 0; i < 4; i++) {
        Serial.println(mux_out_s[i]);
        pinMode(mux_out_s_[i], OUTPUT);
        //delay(10);
    }
    Serial.println("Mux Out Pins Set");
    //Set Mux Pins to LOW
    digitalWrite(mux_read_enable_, LOW);
    digitalWrite(mux_out_enable_, LOW);
    digitalWrite(mux_out_pin_, LOW);

    Serial.println("Mux Read Pins Set to LOW");
    GPIO.out_w1ts = ((uint32_t)0 << mux_read_s_[0]);
    Serial.print("Mux read pin set: ");
    Serial.println(mux_read_s_[0]);
    GPIO.out_w1ts = ((uint32_t)0 << mux_read_s_[1]); //Issue?
    Serial.print("Mux read pin set: ");
    Serial.println(mux_read_s_[1]);
    GPIO.out_w1ts = ((uint32_t)0 << mux_read_s_[2]);//Issue?
    Serial.print("Mux read pin set: ");
    Serial.println(mux_read_s_[2]);
    GPIO.out_w1ts = ((uint32_t)0 << mux_read_s_[3]); //Issue?
    Serial.print("Mux read pin set: ");
    Serial.println(mux_read_s_[3]);
    

    Serial.println("Mux Out Pins Set to LOW");
    GPIO.out_w1ts = ((uint32_t)0 << mux_out_s_[0]);
    Serial.println("Mux out pin: " + String(mux_out_s_[0]) + "set to LOW");
    GPIO.out_w1ts = ((uint32_t)0 << mux_out_s_[1]);
    Serial.println("Mux out pin: " + String(mux_out_s_[1]) + "set to LOW"); //Issue?
    GPIO.out_w1ts = ((uint32_t)0 << mux_out_s_[1]);
    Serial.println("Mux out pin: " + String(mux_out_s_[2]) + "set to LOW");
    GPIO.out_w1ts = ((uint32_t)0 << mux_out_s_[3]);
    Serial.println("Mux out pin: " + String(mux_out_s_[3]) + "set to LOW");
```

### Debug Message

```plain
E (4607) gpio: GPIO can only be used as input mode
[  4613][E][esp32-hal-gpio.c:130] __pinMode(): GPIO config failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-01-22:
The error suggests that you are trying to set IO34+ as OUTPUT. Another thing I notice is `pinMode(mux_read_s_[i], OUTPUT);` . Shouldn't this be INPUT?

Ivanbk, 2024-01-22:
Thanks for the fast reply. IO34 & IO35 are set as INPUT, every other pin as OUTPUT. As for the mux_read_s_ array it should be output, as this array is the address of the 16bit mux. I have now tried:
'gpio_set_direction(GPIO_NUM_XX, GPIO_MODE_INPUT);'
and the error is no longer present, but i cannot seem to write to the pin correctly. No Output pins are responding to write function

closed 2024-01-23 by Ivanbk