espressif/arduino-esp32 #9155: French Keyboard

nano54120, 2024-01-22:
### Board

ESP32-S2-Pico

### Device Description

Esp32s2

### Hardware Configuration

NC

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

How to use libraries/USB/src/USBHIDKeyboard.h as french keyboard ?

### Sketch

```cpp
NC
```

### Debug Message

```plain
NC
```

### Other Steps to Reproduce

NC

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-01-23:
Hi @nano54120, what exact issues you have?

nano54120, 2024-01-23:
Hi @P-R-O-C-H-Y, I use USBHIDKeyboard to send my sentences and sequences like éèçàùô~€ do not pass. I tried to modify the USBHIDKeyboard.cpp file by changing the keyboard table to French but that doesn't work. It seems that symbols with the AltGr key do not pass.
Thanks to Google translate ;-)

nano54120, 2024-01-29:
Ok, i've done.

P-R-O-C-H-Y, 2024-01-30:
> Ok, i've done.

@nano54120 I can't see any sketch posted here.

P-R-O-C-H-Y, 2024-01-30:
You can't use `Keyboard.print()` to print symbols outside the first [128 ASCII characters](https://www.w3schools.com/charsets/ref_html_ascii.asp) also found [HERE](https://github.com/espressif/arduino-esp32/blob/cceebb58f170c34a2356f0d22106dd6f0d141023/libraries/USB/src/USBHIDKeyboard.cpp#L87) in the `USBHIDKeyboard.cpp`.
In order to print those characters you need to emulate the keyboard combination for pressing those.
As example for writing € symbol for US keyboard layout (set in OS - tested on macOS) you need to do this:
```
// Using keyboard print as used only ASCII available 128 chars.
Keyboard.print("Printing euro symbol: ");
// Using combination of keys pressed to get € symbol printed. 
Keyboard.press(KEY_LEFT_SHIFT);
Keyboard.press(KEY_LEFT_ALT);
Keyboard.press('2');
Keyboard.releaseAll();
```

nano54120, 2024-01-30:
Sorry, i must have misunderstood. I posted my message on the discussion 🤣. I understand your explanation. I thought we could do it directly with the library as we can with the Adafruit versions. I will therefore add a conversion table for the symbols concerned in my codes. Thank you so much.

P-R-O-C-H-Y, 2024-01-30:
@nano54120 I am closing it as resolved. If there is a need feel free to reopen :)

closed 2024-01-30 by P-R-O-C-H-Y