espressif/arduino-esp32 #9156: esp32-c6 zigbee ZHA support

ilker-aktuna, 2024-01-22:
### Related area

Zigbee , ZHA

### Hardware specification

esp32-c6 , esp32-h2

### Is your feature request related to a problem?

this is not problem related.

### Describe the solution you'd like

Zigbee support with ZHA (home automation profiles)  would be great. With sone examples.

### Describe alternatives you've considered

none

### Additional context

none

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-01-22:
While I would love to see a more useful example like discovery/query, there is a basic switch ZCZR example in the [PR](https://github.com/espressif/arduino-esp32/pull/9024/files#diff-d5afb99ca40b3aba58fad51571a1acf6579dbc0aa651d334cf5498e86d10a0a3) which is soon to be merged.

ilker-aktuna, 2024-01-23:
Ok. I am trying one of these examples now on my C6 dev module
But it goes into boot loop with following output:
```

22:46:56.277 -> ESP-ROM:esp32c6-20220919
22:46:56.277 -> Build:Sep 19 2022
22:46:56.277 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
22:46:56.277 -> SPIWP:0xee
22:46:56.277 -> mode:DIO, clock div:2
22:46:56.277 -> load:0x4086c410,len:0xc24
22:46:56.277 -> load:0x4086e610,len:0x2708
22:46:56.277 -> load:0x40875728,len:0x594
22:46:56.277 -> entry 0x4086c410
22:46:56.464 -> 
22:46:56.535 -> assert failed: zb_osif_nvram_init zb_esp_nvram.c:70 (zb_partition != NULL)
22:46:56.535 -> Core  0 register dump:
22:46:56.535 -> MEPC    : 0x40801d96  RA      : 0x40807260  SP      : 0x40819c90  GP      : 0x4080e0d0  
22:46:56.535 -> TP      : 0x407e9fd8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
22:46:56.535 -> S0/FP   : 0x00000001  S1      : 0x40819e0a  A0      : 0x40819cf4  A1      : 0x4080e4d5  
22:46:56.535 -> A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40817000  
22:46:56.568 -> A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x0000007d  S3      : 0x40819ce8  
22:46:56.568 -> S4      : 0x40819ce8  S5      : 0x42070f76  S6      : 0x00000000  S7      : 0x420c4a28  
22:46:56.568 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
22:46:56.568 -> T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
22:46:56.601 -> MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
22:46:56.601 -> MHARTID : 0x00000000  
22:46:56.601 -> 
2:46:56.601 -> Stack memory:
22:46:56.601 -> 40819c90: 0x40816a30 0x000000c0 0x420c49d8 0x4080c9c0 0x00000004 0x00000000 0x4081766c 0x4080e4d4
22:46:56.601 -> 40819cb0: 0x00000000 0x00000000 0x4081766c 0x40003037 0x00000000 0x4080e97c 0x420c4a64 0x4080eccc
22:46:56.601 -> 40819cd0: 0x420c4a28 0x4080e98c 0x40819cbc 0x4080e990 0x420c49d8 0x4080e4d4 0x00000000 0x00000000
22:46:56.634 -> 40819cf0: 0x00000000 0x65737361 0x66207472 0x656c6961 0x7a203a64 0x736f5f62 0x6e5f6669 0x6d617276
22:46:56.634 -> 40819d10: 0x696e695f 0x627a2074 0x7073655f 0x72766e5f 0x632e6d61 0x2030373a 0x5f627a28 0x74726170
22:46:56.634 -> 40819d30: 0x6f697469 0x3d21206e 0x4c554e20 0x4000294c 0x4081bb80 0x40880000 0x40816a30 0xffffffff
22:46:56.668 -> 40819d50: 0x420cf000 0x408146cc 0x4080f6cc 0xbaad5678 0x40816a30 0x00000014 0x4087cd4c 0x4080731c
22:46:56.668 -> 40819d70: 0x00000000 0x00000000 0x4087cd4c 0x40807852 0x00000000 0x00000000 0x00000000 0x000000ff
22:46:56.668 -> 40819d90: 0x40880000 0x4081bb80 0x40816a30 0x400225ca 0x40817000 0x4081bb88 0x40816a30 0x4080b79c
22:46:56.668 -> 40819db0: 0x40817000 0x00000000 0x4081bb88 0x679ead0d 0x00000000 0x00000000 0x00000000 0x00000000
22:46:56.701 -> 40819dd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x420cf000 0x408146cc 0x4080f6cc 0x42070f76
22:46:56.701 -> 40819df0: 0x407e9fd8 0x00000000 0x40819e40 0x420167e0 0x00000000 0x00000000 0x40819e40 0x4200761a
22:46:56.701 -> 40819e10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x420000d4
22:46:56.701 -> 40819e30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000002 0x00000000 0x00000006 0x00000bb8
22:46:56.734 -> 40819e50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
22:46:56.734 -> 40819e70: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
22:46:56.734 -> 40819e90: 0xbaad5678 0x00000160 0xabba1234 0x00000154 0x40819de0 0xffffffff 0x40814df0 0x40814df0
22:46:56.734 -> 40819eb0: 0x40819ea0 0x40814de8 0x00000014 0x679ead0d 0x40817000 0x40819ea0 0x00000000 0x00000005
22:46:56.768 -> 40819ed0: 0x40818e90 0x6267695a 0x6d5f6565 0x006e6961 0x00000000 0x00000000 0x40819e80 0x00000005
22:46:56.768 -> 40819ef0: 0x00000000 0x00000000 0x00000000 0x00000000 0x40817458 0x408174c0 0x40817528 0x00000000
22:46:56.768 -> 40819f10: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x420923f0 0x00000000 0x00000000
22:46:56.768 -> 40819f30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
22:46:56.801 -> 40819f50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
22:46:56.801 -> 40819f70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
22:46:56.801 -> 40819f90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
2:46:56.801 -> 40819fb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
22:46:56.834 -> 40819fd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
22:46:56.834 -> 40819ff0: 0x00000000 0xbaad5678 0x0000006c 0xabba1234 0x00000060 0x00000000 0x00000001 0x00000000
22:46:56.834 -> 4081a010: 0x00000000 0x00000000 0x4081a074 0xb33fffff 0x00000000 0xffffffff 0x00000000 0x00000000
22:46:56.834 -> 4081a030: 0x00000000 0x00000000 0x4081a0d8 0x00000000 0x00000000 0xffffffff 0x00000000 0x00000000
22:46:56.869 -> 4081a050: 0x00000000 0x00000000 0x4081a13c 0x00000000 0x00000000 0xbaad5678 0x00000060 0xabba1234
22:46:56.869 -> 4081a070: 0x00000054 0x00000000 0x4081a074 0x00000000 0x00000000 0x00000000 0x4081a08c 0xffffffff
22:46:56.869 -> 
22:46:56.869 -> 
22:46:56.869 -> 
22:46:56.869 -> ELF file SHA256: 561863f77675cea5
22:46:56.869 -> 
22:46:56.869 -> Rebooting...
22:46:56.869 -> ESP-ROM:esp32c6-20220919
[... cut ...]
22:46:56.911 -> mode:DIO, clock div:2
22:46:56.911 -> load:0x4086c410,len:0xc24
22:46:56.911 -> load:0x4086e610,len:0x2708
22:46:56.911 -> load:0x40875728,len:0x594
22:46:56.911 -> entry 0x4086c410
22:46:57.283 -> 
22:46:57.314 -> assert failed: zb_osif_nvram_init zb_esp_nvram.c:70 (zb_partition != NULL)
```

what am I doing wrong ?
the file I'm trying is :
https://raw.githubusercontent.com/espressif/arduino-esp32/bddb847d3b7cb2f4bc5c73a1152c04b5cce4f461/libraries/ESP32/examples/Zigbee/Zigbee_Light_Bulb/Zigbee_Light_Bulb.ino

lbernstone, 2024-01-23:
There's options for Zigbee mode under the tools menu.  You need to select the correct mode.  That will set your [partition scheme](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/zigbee_zczr.csv) as well.

ilker-aktuna, 2024-01-23:
ok. I figured it out now.
Now it compiles and runs.
But this sample doesn't try to join any hub.
How can I modify the code so that it tries to join a zigbee network/hub ?

ilker-aktuna, 2024-01-23:
I changed the "esp_zb_task"  with following function and now my Hub can detect the esp32-c6 but can not complete device initialization.
Besides, I do not see log outputs on serial.

How can I troubleshoot ?

```

static void esp_zb_task(void *pvParameters)
{
    // initialize Zigbee stack with Zigbee end-device config 
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    // set the on-off light device config
    uint8_t test_attr, test_attr2;
 
    test_attr = 0;
    test_attr2 = 4;
    // basic cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr2);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr2);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    // identify cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);
    // group cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &test_attr);
    // scenes cluster create with standard cluster + customized
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);
    esp_zb_cluster_update_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &test_attr);
    // on-off cluster create with standard cluster config
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    // create cluster lists for this endpoint
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    // update basic cluster in the existed cluster list
    //esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    // add created endpoint (cluster_list) to endpoint list
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

```

lbernstone, 2024-01-23:
https://docs.espressif.com/projects/arduino-esp32/en/latest/troubleshooting.html#serial-not-printing

ilker-aktuna, 2024-01-23:
> https://docs.espressif.com/projects/arduino-esp32/en/latest/troubleshooting.html#serial-not-printing

actually my problem is not "serial not printing" , because it prints what I put in setup()
I also tried a serial print on esp_zb_task() , it works.
however, if I put something under zb_action_handler or zb_attribute_handler , they do not show up.

I will try the signal handler now.

ilker-aktuna, 2024-01-23:
ok, serial print works whereever it is triggered.
However, the log_i() or log_e() statements do not work.
why ?

ilker-aktuna, 2024-01-23:
I managed to get logs using serial.println (log_e , log_i did not work)

so when I test, I finally managed to get these: 
00:37:20.151 -> ZDO Config Ready
00:37:20.151 -> ESP_FAIL

so it seems like it has a ZDO config. However, my hub did not add it yet.
Does it store ZDO data somewhere ? Can I delete it to start fresh ?

ilker-aktuna, 2024-01-25:
I also created this issue under zigbee sdk repo:
https://github.com/espressif/esp-zigbee-sdk/issues/228

But nobody seems to be interested there...

Hedda, 2024-10-07:
FYI, looks like a Zigbee API/wrapper for Arduino on ESP32 ([arduino-esp32](https://github.com/espressif/arduino-esp32)) is being working on very activly right now and initial Zigbee library support has been merged into both `master` and `release/v3.1` branches, so initial support will be included in upcoming 3.0.6 and 3.1.0-rc2 releases of arduino-esp32:

* https://github.com/espressif/arduino-esp32/issues/10135

ilker-aktuna, 2024-10-07:
thank you. Still no documentation though. Examples are still limited.

P-R-O-C-H-Y, 2024-11-14:
Hi everyone, I am closing this issue as there are now 8 examples for the new and easy to use Zigbee library. The old examples which were just port from esp-zigbee-sdk were removed. All the examples can be found here: https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples

closed 2024-11-14 by P-R-O-C-H-Y