espressif/arduino-esp32 #9159: Support for Lilygo T-ETH-POE (T-Internet-POE)

PurP13cray0n, 2024-01-22:
### Related area

Board Support for Lilygo T-ETH-POE

### Hardware specification

Support for Lilygo T-Internet-POE

### Is your feature request related to a problem?

I'm facing an issue where I need the Lilygo T-ETH-POE (T-Internet-POE) board to be directly supported here. Platformio (the backend of ESPHome) requires it to be directly supported here so that they can support it.  As a result, ESPHome would then be able to support it.
[https://www.lilygo.cc/products/t-internet-poe](url)
[image]

### Describe the solution you'd like

I would like the Lilygo T-ETH-POE board to be supported as a board type.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

WombatHollow, 2024-01-22:
I would also like support for this board please

VojtechBartoska, 2024-01-30:
Hello, the board needs to be added by Community/Board vendor.

I just quickly searched LilyGo contributors so far: https://github.com/espressif/arduino-esp32/pulls?q=is%3Apr+label%3A%22Type%3A+3rd+party+Boards%22+is%3Aclosed+lilygo

You can try to contact vendor directly.

I am closing this ticket to keep issue tracker managable.

PurP13cray0n, 2024-02-03:
> Hello, the board needs to be added by Community/Board vendor.
>
> I just quickly searched LilyGo contributors so far: [...]

I would be more than happy to help integrate this board the only reason why I posted this here was to see if I had to go further down a rabbit hole of making issues until I find the actual repository that it needs to be added in. If you could direct me in the direction needed for me to be able to add support for this board, that would be great.

VojtechBartoska, 2024-02-05:
@PurP13cray0n It needs to be added to this repository.

Did you take a look on the link I shared above?

Basically you need to now pin layout of that board and you can inspire from any other Board addition.