espressif/arduino-esp32 #9160: ESP 32 failed to connect hosting, while in local server just work fine

ulinnuhaluthfi, 2024-01-23:
### Board

ESP 32 

### Device Description

Devkit V1

### Hardware Configuration

no hardware communication, i just using random value to imitate sensor value

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

9600

### Description

i try to make online my sensor monitoring data to free hosting 000webhostapp, i just simply upload my htdocs file, make it match database name, table and setting connection to database. but it just still doesn't work.
But if i try run the code at local server XAMPP, it just work fine.

### Sketch

```cpp
#include "WiFi.h"
#include "HTTPClient.h"

//variabel untuk set konektifitas hotspot
const char* ssid = "KAPRODI-TEI";  //"mikrofologi";
const char* pass = "tanyapakadib"; //"Nom3rhapeku1";

//variabel host server lokal/ online
const char* host = "monsen32.000webhostapp.com";   //"192.168.1.8";

void setup() {
  Serial.begin(9600);
  randomSeed(analogRead(0));

  //koneksi ke WiFi
  WiFi.begin(ssid, pass);
  Serial.println("Mengubungkan...");
  //Tamilan sermon ketika wifi belum terkoneksi
  while(WiFi.status() != WL_CONNECTED){
    Serial.print("."); delay(500);
  }
  //ketika sudah terkoneksi
  Serial.print("WiFi Terhubung");
}

void loop() {
  float suhu = random(32, 41);
  int kelembaban = random(16, 31);
  int ldr = random(200, 301);
  
  Serial.print("Nilai Sensor suhu: "); Serial.println(suhu);
  Serial.print("Nilai Sensor kelembaban: "); Serial.println(kelembaban);
  Serial.print("Nilai Sensor LDR: "); Serial.println(ldr);
  Serial.println("");

  //kirim data ke server lokal/ online
  WiFiClient client ;  //"client" disini merupakan variabel bebas
  //inisialisasi port web server apache 80
  const int httpPort = 80;
  if( !client.connect(host, httpPort) ){ //jk koneksi ke variabel host melalui port gagal
    Serial.println("Koneksi Gagal");     //maka tampilkan notif "koneksi gagal"
    return;                              //kembali ke loop pertama
  }
  else if(client.connect(host, httpPort)){
    Serial.println("Koneksi HTTP Berhasil");
  }

  //kondisi sudah bisa terhubung dengan server
  //maka kirim data sensor ke database

  String Link;      //variabel link lokal
  HTTPClient http;  //

  //Link = "http://" + String(host) + "/multisensor/kirimdata.php?suhu=" + String(suhu) + "&kelembaban=" + String (kelembaban) + "&ldr=" + String (ldr);
  Link = "https://" + String(host) + "/kirimdata.php?suhu=" + String(suhu) + "&kelembaban=" + String (kelembaban) + "&ldr=" + String (ldr);

  //eksekusi alamat link yang ada
  http.begin(Link);
  http.GET();

  //baca respon setelah berhasil kirim nilai sensor
  String respon = http.getString();
  Serial.println(respon);
  http.end();

  delay(1000);
}
```

### Debug Message

```plain
Nilai Sensor suhu: 35.00
Nilai Sensor kelembaban: 20
Nilai Sensor LDR: 259

//this is serial monitor print, if the connection work, there is response "sent successfully"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-01-23:
you probably should do `http.POST()` instead

VojtechBartoska, 2024-09-10:
Hello,

due to no answer, I am closing this ticker as expired.

If needed, you can reopen it.

Thank you for understanding.

closed 2024-09-10 by VojtechBartoska