espressif/arduino-esp32 #9175: PlatformIO: platform = https://github.com/espressif/arduino-esp32.git#idf-release/v5.1 errors MissingPackageManifestError: Could not find one of 'platform.json' manifest files in the package

zackees, 2024-01-25:
### Board

Esp32C3

### Device Description

This is a toolchain error, hardware configuration N/A

Example platformio.ini is provided below

### Hardware Configuration

This is a toolchain error, hardware configuration N/A

### Version

other

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I would like to test out the new 5.1 toolchain because it has a fix for the I2S controller during light sleep.

https://github.com/espressif/arduino-esp32/issues/8207

I do not want to re-write my code off of esp32-arduino yet.

### Sketch

```cpp
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32c3]
platform = https://github.com/espressif/arduino-esp32.git#idf-release/v5.1
board = seeed_xiao_esp32s3
framework = arduino

upload_speed = 921600

; change microcontroller
board_build.mcu = esp32c3
;upload_speed = 115200

; change MCU frequency
; board_build.f_cpu = 160000000L

; change WiFi firmware
board_build.variant = esp32c3

monitor_filters = 
	default
	esp32_exception_decoder

monitor_speed = 115200

```
```

### Debug Message

```plain
Resolving esp32s3 dependencies...
Platform Manager: Installing git+https://github.com/espressif/arduino-esp32.git#idf-release/v5.1
git version 2.42.0.windows.2
Cloning into 'C:\Users\niteris\.platformio\.cache\tmp\pkg-installing-rfdo4307'...

MissingPackageManifestError: Could not find one of 'platform.json' manifest files in the package
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-01-25:
@zackees The setup you tried is not working. A working setup to use latest version from branch master is possible by using this PR https://github.com/platformio/platform-espressif32/pull/1281

zackees, 2024-01-27:
Confirmed the pull request worked.

...now I have to port ledc, i2s and the watchdog timer...

closed 2024-01-27 by Jason2866