espressif/arduino-esp32 #9180: will this library support esp32 p4?

aaronwu2017, 2024-01-27:
### Related area

Question

### Hardware specification

esp32 p4

### Is your feature request related to a problem?

Do you have a timeline/roadmap for esp32 p4 support?

### Describe the solution you'd like

answers to my questions

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-01-27:
Timelines are never given here for new chips.  Support will not happen before the ESP-IDF version which fully supports a new chip is released.  Typical lag times are then 6 months to a year to make their way into arduino-esp32.

aaronwu2017, 2024-01-27:
OK thanks

closed 2024-01-27 by aaronwu2017