espressif/arduino-esp32 #9183: BLE connect fails after light sleep

flynny75, 2024-01-28:
### Board

Adafruit ESP32 S3

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.14

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

BLE clients will not connect to a server after waking from light sleep

### Sketch

```cpp
e board I am using is an adafruit S3 No PSRAM.

I have included the sketch below:

#include <Arduino.h>
#include "NimBLEDevice.h"
#include <FastLED.h>
#include "WiFi.h"

bool isServer = false;

NimBLEClient* client;
NimBLEAdvertisedDevice* server;

int connectCount = 0;
CRGB leds[1];

class AdvertisedDeviceCallbacks: public NimBLEAdvertisedDeviceCallbacks {
    public:
    void onResult(NimBLEAdvertisedDevice* advertisedDevice) {
        if(advertisedDevice->getName() == "NimBLE_light_sleep")
        {
            server = advertisedDevice;
            NimBLEDevice::getScan()->stop();
        }
    };
};

void flashLed(CRGB colour, int timeOn, int timeOff, int times)
{    
    leds[0] = CRGB::Black;
    FastLED.show();
    vTaskDelay(timeOff / portTICK_PERIOD_MS);

    int i = 0;
    while(i < times)
    {
        leds[0] = colour;
        FastLED.show();
        vTaskDelay(timeOn / portTICK_PERIOD_MS);

        leds[0] = CRGB::Black;
        FastLED.show();
        vTaskDelay(timeOff / portTICK_PERIOD_MS);
        i++;
    }
}

void setup() {
    FastLED.addLeds<NEOPIXEL, 33>(leds, 1);

  if(isServer)
  {
        NimBLEDevice::init("NimBLE_light_sleep");
        auto bleServer = NimBLEDevice::createServer();
        auto advertising = NimBLEDevice::getAdvertising();
        auto service = bleServer->createService("1111");
        auto c = service->createCharacteristic("2222");
        c->setValue("3333");

        service->start();
        advertising->start();

        Serial.begin(115200);
  }
  else
  {
        pinMode(LED_BUILTIN, OUTPUT);
        NimBLEDevice::init("");
        auto scanner = NimBLEDevice::getScan();
        scanner->setAdvertisedDeviceCallbacks(new AdvertisedDeviceCallbacks());
        scanner->start(0, [](NimBLEScanResults r){});
        client = NimBLEDevice::createClient();
        client->setConnectTimeout(10);
  }
}

void loop() {
  if(isServer)
  {
    // N/A
  }
  else
  {
    if(server == NULL){
        vTaskDelay(2000 / portTICK_PERIOD_MS);
        return;
    }

    // Cycle for 5 times without light sleeping
    if(connectCount > 2)
    {
        log_e("===== LIGHT SLEEP =====");

        // light sleep for 5 seconds, user disconects serial
        auto configureLightSleepTimer = esp_sleep_enable_timer_wakeup(5 * 1000000);
        if(configureLightSleepTimer != ESP_OK)
        {
            log_e("bad sleep timer wake enable: %s\n", esp_err_to_name(configureLightSleepTimer));
            flashLed(CRGB::Red, 20, 20, 50);
        }

        auto sleepResult = esp_light_sleep_start();
        if(sleepResult != ESP_OK){
            log_e("bad sleep: %s\n", esp_err_to_name(sleepResult));
            flashLed(CRGB::Red, 20, 20, 50);
        }

        digitalWrite(LED_BUILTIN, HIGH);
        //wake up, pause for 5s to reconnect serial
        vTaskDelay(5000 / portTICK_PERIOD_MS);
        digitalWrite(LED_BUILTIN, LOW);
    }
    else
    {
        log_e("===== DELAY =====");
        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }

    if(client->connect(server))
    {
        flashLed(CRGB::Green, 100, 100, 5);
    }
    else
    {
        flashLed(CRGB::Red, 100, 100, 5);
    }

    vTaskDelay(2000 / portTICK_PERIOD_MS);
    client->disconnect();
    vTaskDelay(2000 / portTICK_PERIOD_MS);

    connectCount++;
  }
}
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Sample sketch requires two boards, one with `isServer = true`, and the other with `isServer = false`. Client will attempt to connect and disconnect twice using vTaskDelay to simulate the pause. It then goes to light sleep. The connection fails to be established after waking from light sleep

https://github.com/h2zero/NimBLE-Arduino/issues/616
https://github.com/platformio/platform-espressif32/issues/1273

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-15 by Parsaabasi

csrutil, 2026-01-01:
Hey @flynny75, have you addressed this issue or found a workaround for the bug? Thanks!