espressif/arduino-esp32 #9184: strptime doesn't parse AM/PM (%p)

chconnor, 2024-01-28:
### Board

ESP32

### Device Description

ESP32-DevKitC-VIE

### Hardware Configuration

many things

### Version

v2.0.11

### IDE Name

vscode + Platform IO

### Operating System

Linux

### Flash frequency

default?

### PSRAM enabled

yes

### Upload speed

default?

### Description

Hopefully this is the proper place to post this -- if not please let me know:

strptime() does not seem to parse AM/PM correctly (using the %p token). You will see in the demo code that it fails to detect AM and always makes the time PM. On my desktop, that demo code will print "hour is 8" but on the ESP32 it will print "hour is 20".

My understanding is that the parsing of %p depends on the locale, which may not use AM/PM, but I don't see a way to configure things in a way that makes %p work.

### Sketch

```cpp
#include <Arduino.h>
#include <stdio.h>
#include <stdlib.h>
#include <locale.h>
#include <time.h>

void setup()
{
  Serial.begin(115200);
  while (!Serial) delay(10);

  char timestr[7] = "8:00AM";
  static struct tm temptm;
  static time_t temptime;
  temptime = time(NULL);
  temptm = *localtime(&temptime);
  strptime(timestr, "%I:%M%p", &temptm);
  Serial.printf("hour is %d\n", temptm.tm_hour);
}

void loop()
{
    delay(100);
}
```

### Debug Message

```plain
n/a
```

### Other Steps to Reproduce

See also this user report: https://esp32.com/viewtopic.php?t=36652

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-01-29:
Hello @chconnor, I can confirm the issue and I have reported it to the related team to take a look at this as it cannot be fixed in Arduino project. Tested also with ESP-IDF with same results. When I know any news I will post an update here.

Lapshin, 2024-02-20:
Looks like there is a bug in newlib library. Sent a fix https://sourceware.org/pipermail/newlib/2024/021031.html

Next IDF's toolchain release will have the fix

VojtechBartoska, 2024-04-23:
this was fixed in esp-4.3.0 of newlib, closing as solved.

closed 2024-04-23 by VojtechBartoska

codegrue, 2024-05-03:
When will this fix be in the released ESP32 Arduino library? I am seeing 6.5.0 as the latest currently.