espressif/arduino-esp32 #9185: COMPILED WITH 5.1?

sboicu, 2024-01-28:
### Board

esp32

### Device Description

n.a.

### Hardware Configuration

n.a.

### Version

latest master (checkout manually)

### IDE Name

Eclipse

### Operating System

MacOS

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

n.a.

### Description

I try to use in eclipse the arduino as component and here is the error I get:

```
esp-idf/esp-idf-v4.4.6/tools/cmake/build.cmake:463 (message):
  HINT: Please check manifest file of the following component(s): arduino

  ERROR: Because project depends on idf (>=5.1) which doesn't match any
  versions, version solving failed.
```

does this mean that thee component is acutally compiled with 5.1 and I shall update my esp-idf?

I followed https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html and there is specifically said that latest Arduino core is based on 4.4.6

any idea?

### Sketch

```cpp
sample code
```

### Debug Message

```plain
na.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-01-28:
Yes, Arduino master is based on IDF 5.1

lbernstone, 2024-01-28:
2.0.X were compiled with IDF 4.X
3.0.X will be compiled with IDF 5.X

sboicu, 2024-01-29:
> 2.0.X were compiled with IDF 4.X
> 3.0.X will be compiled with IDF 5.X

This is what i also have read, but, i have installed 2.0.14 and to me i got the error displayed over there. Maybe in Arduino IDE it is not showing as an error because it is standalone but in Eclipse because is added on a 4.4.6 environment i got this error.
I will double check the installed version of the component and I will report back.

closed 2024-01-29 by sboicu