espressif/arduino-esp32 #9187: Wifi Mode

kayleejacob1, 2024-01-29:
### Board

ESP all Series 

### Device Description

customer board

### Hardware Configuration

normal

### Version

v2.0.14

### IDE Name

latestes

### Operating System

windows 10

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

If wifi.mode is set to to any mode like WiFi.mode(WIFI_AP_STA);

Unable to connect

for regular WiFi.mode(WIFI_STA); it will not connect to any networks

is this nolonger needed if so when did this change.  

### Sketch

```cpp
WiFi.macAddress(macs);
    WiFi.mode(WIFI_AP_STA);
    //// You can remove the password parameter if you want the AP to be open.

    WiFi.softAP(AP, passwordAP);
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-01-29:
Turn on verbose debugging and post a log.

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

peterwillcn, 2026-06-28:
WiFi 7 won't connect