espressif/arduino-esp32 #9191: Bus errors transmitting a specific TWAI message on 33.3kbps CAN

zalexzperez, 2024-01-30:
### Board

ESP32-S3-WROOM-1-N8

### Device Description

Custom board developed by myself for an automotive application. 
Main network is GMLAN's low speed single wire CAN (33.3kbps) using a NCV7356 SWCAN transceiver, from now on, referred to as "LSCAN transceiver"

### Hardware Configuration

The attached sketch only uses the LSCAN transceiver on pins 18 (RX) and 8 (TX). 
It operates in normal mode by setting pins 7 (M0) and 15 (M1) high.

[image]

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Transmitting CAN messages on the low-speed GMLAN (33.3kbps) is successful with both device connected and unconnected to the car. 
There are three diagnostic messages in my car to achieve needle sweep (moving needles to their max position and quickly return to the origin).
Two of them work every time, with no bus errors. However, I'm having errors when attempting to transmit the fuel needle message, the data bytes of which I named as "fuel_meter_sweep [8]" and the function to transmit the message is "fuel_needle_sweep()"

The message should be transmitted every 9 seconds. On the first transmission, the message regularly gets a transmission failed alert and doesn't do anything. Sometimes it shows transmission successful but gets ignored by the car, and rarely, the transmission is successful and does move the fuel needle, but still generating bus errors  along the way that end up causing 32 bus errors, making the TWAI driver go into the BUS OFF state.

Scope view when transmitting the fuel message with the **board disconnected from the car**:
[image]

Scope view when transmitting the fuel message with the board **connected to the car**. The message keeps getting sent every few milliseconds for at least the 140ms that the scope captured. Notice the blanks in between messages, I have no idea what it is:
[image]

The message identifier and data bytes are 100% correct. I can transmit this message using a regular ESP32 (FireBeetle 2 ESP32-E) with Sandeep Mistry's [Arduino CAN](https://github.com/sandeepmistry/arduino-CAN) library, which is unmaintained and doesn't work with the newer ESP32 chips. Also worth mentioning that the FireBeetle board also fails transmitting the message with the native TWAI driver.

Here's the scope view of the fuel message transmitted using the aforementioned Arduino CAN library with the Firebeetle **connected to the car**:
[image]

### Sketch

```cpp
#include <Arduino.h>
#include "driver/twai.h" // CAN Bus
#include "CAN_messages.h"

#define DEBUG

#ifdef DEBUG
#define DEBUGPRINT(X) Serial.print(X)
[... 486 lines cut ...]
Bus error count: 26
Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
Bus error count: 27
Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
Bus error count: 28
Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
Bus error count: 29
Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
Bus error count: 30
Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
Bus error count: 31
Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
Bus error count: 32
TWAI state  -----------------------------> BUS OFF
```

### Other Steps to Reproduce

As I mentioned on the issue description, transmitting the conflictive message works fine with Sandeep's Mistry Arduino-CAN library. 
So I'm attaching the test project I used with his library. The library I used is actually a [forked version](https://github.com/mister0wl/arduino-CAN) that allows working at lower baudrates (the first ESP32 chip versions weren't capable of going below 50kbps, but newer revisions decreased this limit). I modified the ESP32SJA1000.cpp file to include the 33.3kbps baudrate configuration registers.
[ESP32 package transmit test.zip](https://github.com/espressif/arduino-esp32/files/14103813/ESP32.package.transmit.test.zip)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-01-31:
I suggest you poke the IDF team in their repo. We do not provide our own TWAI driver and there they will be able to help you better.

zalexzperez, 2024-01-31:
Well, turns out it was a timing configuration problem. 
These work well:

#define TWAI_TIMING_CONFIG_33k3_KBITS()                                       \
{                                                                          \
  .brp = 120, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false \
}

closed 2024-01-31 by zalexzperez

04ColyZQ8, 2025-11-20:
Sweet!!

04ColyZQ8, 2025-11-21:
I tried it works for 11 bit id but yet working for 29bit extended

zalexzperez, 2025-11-21:
Those are frames that I sniffed from Opcom, not my invention. 
And those bytes at the end are the so called padding bytes, they most likely could be any value.