espressif/arduino-esp32 #9192: Pwm in lightsleep

doublep21, 2024-01-30:
### Related area

gpio pwm

### Hardware specification

esp32-s3

### Is your feature request related to a problem?

pwm don't work in light sleep

### Describe the solution you'd like

pwm using upl or another workaround 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-01-31:
not sure what should be our action here? You are free to use the ULP if that will help you cause.

doublep21, 2024-02-03:
> not sure what should be our action here? You are free to use the ULP if that will help you cause.

I cant use ulp because is not enabled by default , i've tryied to compile arduino core myself but i failed , it compiled but then when flashed resulting in a core panic store prohibited : 
command i've used
`git clone -b  release/v4.4 --recursive https://github.com/espressif/esp-idf.git 
cd esp-idf
./[install.sh](http://install.sh/) esp32s3
. ./[export.sh](http://export.sh/)
cd
git clone -b release/v4.4  https://github.com/espressif/esp32-arduino-lib-builder
 cd esp32-arduino-lib-builder/
sudo ./[build.sh](http://build.sh/) -t esp32s3 -b menuconfig

 ./[build.sh](http://build.sh/) -t esp32s3 -A idf-release/v4.4 -I release/v4.4`

doublep21, 2024-02-03:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42081c22
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1654
load:0x403b6000,len:0xed4
load:0x403ba000,len:0x32dc
entry 0x403b632c
I (29) boot: ESP-IDF v4.4.1-405-g6c5fb29c2c 2nd stage bootloader
I (29) boot: compile time 04:21:58
I (29) boot: chip revision: 0
I (32) qio_mode: Enabling QIO for flash chip WinBond
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : QIO
I (47) boot.esp32s3: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (76) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (83) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (91) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (98) boot:  4 uf2              factory app      00 00 002d0000 00040000
I (106) boot:  5 ffat             Unknown data     01 81 00310000 000f0000
I (113) boot: End of partition table
I (618) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=42a44h (272964) map
I (660) esp_image: segment 1: paddr=00052a6c vaddr=3fc987e0 size=04aach ( 19116) load
I (664) esp_image: segment 2: paddr=00057520 vaddr=40374000 size=08af8h ( 35576) load
I (672) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=8337ch (537468) map
I (756) esp_image: segment 4: paddr=000e33a4 vaddr=4037caf8 size=0bcdch ( 48348) load
I (765) esp_image: segment 5: paddr=000ef088 vaddr=600fe000 size=00024h (    36) load
I (774) boot: Loaded app from partition at offset 0x10000
I (774) boot: Disabling RNG early entropy source...
E (793) esp_core_dump_flash: No core dump partition found!
E (793) esp_core_dump_flash: No core dump partition found!
[  1015][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4203867f  PS      : 0x00060a30  A0      : 0x82038754  A1      : 0x3fcebd40  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x0000000b
A6      : 0x000000ff  A7      : 0x00000001  A8      : 0x08000000  A9      : 0x3fcebd10
A10     : 0x3fca21bc  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00004000  
A14     : 0x00000000  A15     : 0x3fc9aad4  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000010  LBEG    : 0x4206736c  LEND    : 0x4206737b  LCOUNT  : 0x00000000  

Backtrace: 0x4203867c:0x3fcebd40 0x42038751:0x3fcebd70 0x4203875c:0x3fcebd90 0x42001922:0x3fcebdb0 0x4203afa2:0x3fcebe00

  #0  0x4203867c:0x3fcebd40 in TFT_eSPI::begin_tft_write() at lib/TFT_eSPI/TFT_eSPI.cpp:81
      (inlined by) TFT_eSPI::writecommand(unsigned char) at lib/TFT_eSPI/TFT_eSPI.cpp:982
  #1  0x42038751:0x3fcebd70 in TFT_eSPI::init(unsigned char) at lib/TFT_eSPI/TFT_eSPI.cpp:692
  #2  0x4203875c:0x3fcebd90 in TFT_eSPI::begin(unsigned char) at lib/TFT_eSPI/TFT_eSPI.cpp:603
  #3  0x42001922:0x3fcebdb0 in setup() at src/main.cpp:165
  #4  0x4203afa2:0x3fcebe00 in loopTask(void*) at C:/Users/x128b/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: b4610fc4f351646b

E (598) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...

lbernstone, 2024-02-03:
ulp_fsm is not included for -S2 and -S3.  Does that issue go in lib-builder @me-no-dev ?

me-no-dev, 2024-02-04:
> Does that issue go in lib-builder?

Probably yes, if all that is needed has to be enabled there. Can fsm_ulp and riscv_ulp work together? if not, then the RV is preferred.

VojtechBartoska, 2024-09-12:
assigning to 3.0.5 milestone, lets resolve this ticket.

closed 2025-01-15 by Parsaabasi