espressif/arduino-esp32 #9193: TWAI V2 update is Arduino IDE

MatheusCaltran, 2024-01-31:
### Related area

Update to TWAI

### Hardware specification

ESP32

### Is your feature request related to a problem?

I can't get Multiple TWAI Instances to work on Arduino IDE. 

### Describe the solution you'd like

update the TWAI for V2 in arduino IDE

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-01-31:
Arduino based on IDF 5.3 is coming later this year. Next in line is 5.1 to be relased

MatheusCaltran, 2024-01-31:
Do you have any predictions on the arrival of 5.1 and 5.2?

me-no-dev, 2024-02-01:
5.1 is in alpha and soon to be in RC. 5.3 have not yet started on.

closed 2024-02-01 by MatheusCaltran