espressif/arduino-esp32 #9203: HTTP Status code is sometimes equal to -11 after successful post request.

ms-somanna, 2024-02-02:
### Board

ESP32 Dev Board

### Device Description

Custom Dev Board

### Hardware Configuration

The following modules are attached to the board.
- Rs485 to TTL converter module
- RTC module
- LTE module

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm fetching energy meter data from the RS485 module, converting it to Json objects and sending the Json data as payload in a http post request to Google Sheets. I'm using the HTTP status code returned by `HTTP.GET()` function to check if the request was successful or not. The problem is sometimes even after successfully adding the json data to google sheets, the http status code returned is -11. This obviously leads to assuming the request failed even though it succeeded.

### Sketch

```cpp
String url = "https://script.google.com/macros/s/"+GOOGLE_SCRIPT_ID+"/exec";
Serial.println(url);
Serial.println("Making a request");
http.begin(url, root_ca);     //Specify the URL and certificate
http.addHeader("Content-Type", "application/json");
httpCode = http.POST("[" + payload + "]"); 
http.end();
```

### Debug Message

```plain
No error/debug messages were captured till now. I haven't yet made the code capture the HTTP response string either. Please do let me know if you require it.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-02-02:
You can use HTTPClient::errorToString() to turn that into a text response.  In this case it is a read timeout, most likely when waiting for the server to give you a response.  Google can certainly be slow.  Use setTimeout to increase the timeout, default is 5000 (ms).

VojtechBartoska, 2024-09-12:
looks this can be closed as answered, if needed, you can reopen. Thanks

closed 2024-09-12 by VojtechBartoska