espressif/arduino-esp32 #9209: ERROR - Connect ESP32 to Emstat Pico module

OrelBatito, 2024-02-05:
### Board

ESP32

### Device Description

ESP32 Dev Module 

### Hardware Configuration

[Scheme Of Our Connection Pico-ESP32.pdf](https://github.com/espressif/arduino-esp32/files/14164162/Scheme.Of.Our.Connection.Pico-ESP32.pdf)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.2.1

### Operating System

Windows 11

### Flash frequency

80-240 MHz

### PSRAM enabled

no

### Upload speed

230400

### Description

Mention that when we tried to run the following code with an Arduino Uno board, the code verification succeeded:
[Arduino_uno_ToPico.txt](https://github.com/espressif/arduino-esp32/files/14164615/Arduino_uno_ToPico.txt)

BUT, this code verification failed when we tried to run it with an ESP32 Dev Module board. We received the following note:

"c:\Users\1000o\OneDrive\Documents\Arduino\libraries\MethodSCRIPTComm\MSComm.c:253:14: error: conflicting types for 'GetUnitPrefixValue'
 const double GetUnitPrefixValue(char charPrefix)
              ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\1000o\OneDrive\Documents\Arduino\libraries\MethodSCRIPTComm\MSComm.c:38:
c:\Users\1000o\OneDrive\Documents\Arduino\libraries\MethodSCRIPTComm\MSComm.h:366:8: note: previous declaration of 'GetUnitPrefixValue' was here
 double GetUnitPrefixValue(char charPrefix);
        ^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1"

We believe that this error occurred because the libraries and the main code are not compatible with the ESP32, but we have not been successful in resolving this issue.

### Sketch

```cpp
The code provided in the following link:
https://github.com/PalmSens/MethodSCRIPT_Examples/tree/master/MethodSCRIPTExample_Arduino

There are 2 folders: 'MethodSCRIPTComm' and 'MathHelperLibrary' that we installed as libraries in Arduino IDE 2.2.1. We used the 'MethodSCRIPTExample' and tried to run the code with an ESP32.
```

### Debug Message

```plain
c:\Users\1000o\OneDrive\Documents\Arduino\libraries\MethodSCRIPTComm\MSComm.c:253:14: error: conflicting types for 'GetUnitPrefixValue'
 const double GetUnitPrefixValue(char charPrefix)
              ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\1000o\OneDrive\Documents\Arduino\libraries\MethodSCRIPTComm\MSComm.c:38:
c:\Users\1000o\OneDrive\Documents\Arduino\libraries\MethodSCRIPTComm\MSComm.h:366:8: note: previous declaration of 'GetUnitPrefixValue' was here
 double GetUnitPrefixValue(char charPrefix);
        ^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

### Other Steps to Reproduce

Emstat Pico Module Datasheet:
[EmStat Pico Brochure.pdf](https://github.com/espressif/arduino-esp32/files/14164697/EmStat.Pico.Brochure.pdf)
[PSDAT-ESP-EmStat-Pico-Datasheet.pdf](https://github.com/espressif/arduino-esp32/files/14164698/PSDAT-ESP-EmStat-Pico-Datasheet.pdf)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-02-05:
This is an error in the 3rd party library.  In the .h file, it is defined as a double, in the .c it is redefined as const double.  Change the header to match.

OrelBatito, 2024-02-06:
Thank you! We fixed this error, but we received the following ERROR:
"A fatal error occurred: Failed to connect to ESP32: Download mode successfully detected, but no sync reply was received: The serial TX path seems to be down.
For troubleshooting steps, visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2"
Despite the connections being as shown in the scheme file and double-checking our board, the error persists.

lbernstone, 2024-02-06:
esp32 expects to be the only device on the uart line while uploading firmware.  You have made a poor choice to use those same pins for another device.  There are many other pins that could be used for a uart connection to your pico.

VojtechBartoska, 2024-09-12:
Hello, there is nothing to be fixed on Arduino core side, closing this ticket as wont do. Thanks for understanding.

closed 2024-09-12 by VojtechBartoska