espressif/arduino-esp32 #9218: SD card implementation does not set SPI mode before activating card

and3rson, 2024-02-06:
### Board

ESP32-C3

### Device Description

ESP32-C3 SuperMini with ST7789 display and micro SD Card on the same SPI bus.

### Hardware Configuration

Several pushbuttons on different pins (unrelated to the problem)

### Version

v2.0.14

### IDE Name

nvim

### Operating System

ArchLinux

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

N/A

### Description

SD library sets SPI mode to 0 as expected.

However, if SPI bus config changes between SD calls (e. g. a TFT display sets mode to 3), the SD library will attempt to use the incorrect mode, fail, and re-init the card.

Possible solution is to ensure that SPI mode is valid when enabling the card, and set it to 0 otherwise.

### Sketch

```cpp
Arduino_ST7789 gfx(...); // https://github.com/moononournation/Arduino_GFX

// ...

SD.open("/").close();
gfx.fillScreen(display.color565(0, 0, 0));
SD.open("/").close(); // Triggers the error message
```

### Debug Message

```plain
[  6140][E][sd_diskio.cpp:200] sdCommand(): Card Failed! cmd: 0x0d
[  6141][E][sd_diskio.cpp:622] ff_sd_status(): Check status failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-02-07:
This is a bit hard to believe, because SD access is guarded by SPI transactions that do set the mode to 0: https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L468

If what you say indeed happens, then you must have hit some edge case/bug that makes transactions not working.
@P-R-O-C-H-Y PTAL

and3rson, 2024-02-07:
I will try to make a minimum reproduction project where it's reproduced 100% of the time and post it shortly.

and3rson, 2024-02-07:
I also noticed another curious (unrelated) thing: clock polarity is not affected by `beginTransaction` until the first right. Is this the expected behavior? This has affected me in another project.

and3rson, 2024-02-07:
Update: my bad, this was only reproducible with v2.0.14. Turns out my project was being built with v2.0.6 for some reason.

I cannot reproduce this anymore with v2.0.14.

closed 2024-02-07 by and3rson