espressif/arduino-esp32 #9221: `SPI.beginTransaction` does not set SCK value until the first write

and3rson, 2024-02-07:
### Board

ESP32-C3

### Device Description

No hardware

### Hardware Configuration

SCK = 0
MISO = 8
MOSI = 1

### Version

latest master (checkout manually)

### IDE Name

n/a

### Operating System

n/a

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

n/a

### Description

SPI clock polarity is not affected by beginTransaction until the first write. This leads to slave device ignoring some SPI packets.

```This has affected my TFT display, since clock polarity was not being changed before activating the display until the first command was sent.
The only solution was to write a dummy byte directly after beginTransaction.
Related issue: https://github.com/moononournation/Arduino_GFX/issues/433

### Sketch

```cpp
SPI.beginTransaction(SPISettings(..., ..., SPI_MODE0)); // Set SPI mode to 0 (CPOL=0)
    SPI.write(0xFF);
    SPI.endTransaction();
    // SCK is now LOW (SPI mode 0)
    SPI.beginTransaction(SPISettings(..., ..., SPI_MODE3)); // Set SPI mode to 3 (CPOL=1)
    SPI.endTransaction();
    // Problem: SCK is STILL LOW (SPI despite being in mode 3).
    // If we assert device's SS signal, it will enter incosistent state
    // due to SCK being low.
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

and3rson, 2024-02-07:
Here's a better example to reproduce this:

```cpp
#include <Arduino.h>
#include <SPI.h>

#define SD_SCK 0
#define SD_MISO 8
#define SD_MOSI 1
#define DEV_CS 5

#define SPI_SPEED 100000 // 100 KHz

void setup() {
    SPI.begin(SD_SCK, SD_MISO, SD_MOSI);
    pinMode(DEV_CS, OUTPUT);
    digitalWrite(5, HIGH);

        delay(1000); // all delays are indented for readability

    SPI.beginTransaction(SPISettings(SPI_SPEED, MSBFIRST, SPI_MODE0));
        delayMicroseconds(10);
    digitalWrite(DEV_CS, LOW);
        delayMicroseconds(10);
    SPI.write(0xAA);
        delayMicroseconds(10);
    digitalWrite(DEV_CS, HIGH);
        delayMicroseconds(10);
    SPI.endTransaction();

    delayMicroseconds(20);

    SPI.beginTransaction(SPISettings(SPI_SPEED, MSBFIRST, SPI_MODE3));
        delayMicroseconds(10);
    digitalWrite(DEV_CS, LOW);
        delayMicroseconds(10);
    SPI.write(0xAA);
        delayMicroseconds(10);
    digitalWrite(DEV_CS, HIGH);
        delayMicroseconds(10);
    SPI.endTransaction();
}

void loop() {}
```

Actual results:
[image]

and3rson, 2024-02-07:
Possible workaround is to use hardware CS:

```cpp
    SPI.begin(SD_SCK, SD_MISO, SD_MOSI, DEV_CS);
    SPI.setHwCs(true);
```

This way, CS is properly asserted AFTER clock polarity changes:
[image]

However, this makes it impossible to have multiple devices share the bus.

SuGlider, 2024-02-09:
@me-no-dev - Please take a look. Thanks!

P-R-O-C-H-Y, 2024-02-14:
I am taking a look what we can do about it :)

and3rson, 2024-02-14:
FYI - the workaround that I suggested (and which was applied) to https://github.com/moononournation/Arduino_GFX was to write a dummy byte after `beginTransaction` (as outlined [here](https://github.com/moononournation/Arduino_GFX/issues/433).

and3rson, 2024-03-03:
FWIW, the easiest solution for this issue is by writing a dummy byte within the `AcquireSPI` guard, right after https://github.com/espressif/arduino-esp32/blob/aed7b4fa598765db61678d3253401b8cd64a8ec8/libraries/SD/src/sd_diskio.cpp#L474:

```cpp
    explicit AcquireSPI(ardu_sdcard_t* card)
        : card(card)
    {
        card->spi->beginTransaction(SPISettings(card->frequency, MSBFIRST, SPI_MODE0));
        card->spi->write(0xFF); // Force SCK to settle in correct polarity
    }
```

and3rson, 2024-03-04:
I just did a test and added this code at the end of `spiTransaction`:

cpp
    spi->dev->cmd.update = 1;
    while (spi->dev->cmd.update);
```

...and it fixed the clock polarity. My assumption is that setting `update` triggers the hardware to update underlying SPI hardware state.

Before fix:

![зображення](https://github.com/espressif/arduino-esp32/assets/190107/877c4e4c-05ed-41ee-bdee-d6b17e0de8a1)

After fix:

![зображення](https://github.com/espressif/arduino-esp32/assets/190107/fcaeb62f-26f2-4fb9-a204-cd85edf662fa)

Jason2866, 2024-03-04:
@and3rson Does the faulty behaviour affects only the C3 and S3?
The PR you provide to fix is applied to this two MCU variants only.

and3rson, 2024-03-04:
@Jason2866 Those were the inly ones I have and was able to test... Not sure about other models.

Jason2866, 2024-03-04:
So there should be more tests done, if the other MCUs does have this issue too.

P-R-O-C-H-Y, 2024-03-04:
I will test all chips tomorrow. Also this PR may also help with troubles I had adding multiple CS pins support.

closed 2024-03-05 by me-no-dev

and3rson, 2024-12-09:
@SuGlider @P-R-O-C-H-Y
Is it possible to backport this into v2.x.x branch?
We have released & shipped a device that really needs this fix in 2.x.x, and we cannot upgrade to v3.x.x due to large amounts of existing code.
Thanks!

me-no-dev, 2024-12-09:
@and3rson we no longer support 2.x, but the fix is only two lines, so you can add it to your codebase and use it. https://github.com/espressif/arduino-esp32/pull/9333/files

and3rson, 2024-12-09:
@me-no-dev Thanks for quick response!
There's a big issue with this approach though: we'll need to make a fork and somehow ensure everyone (including platformio package) uses our fork just for this one fix. So we'll also need to fork platformio.
Is it somehow still possible for you to release this as a minor version of 2.x.x? It would really help us a lot. Of course, I'll completely understand if you decide not to.
Thanks for your time!

me-no-dev, 2024-12-09:
@and3rson we have updated/fixed many things in our release scripts since and I am afraid that there will be issues if we try to release a new 2.x version. We are a small team and we can not afford to break the 3.x releases. Also I suspect that PIO Labs will not push that new release to their servers anymore. What we can do is accept PR with the change to our `release/v2.x` branch and then I believe that there is a way to instruct PlatformIO to use the github version of the core. Would something like that work for you?

Jason2866, 2024-12-09:
@and3rson Imho the way suggested from @me-no-dev is a good solution.
With adding this after the `platform = espressif32 @ <version>`
```
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/archive/refs/heads/release/v2.x.zip
```
the branch `release/v2.x` is used.

and3rson, 2024-12-09:
@me-no-dev @Jason2866 This would be perfect, thank you! I'll submit a PR shortly.

PR #9333 (2024-03-05): Fix spiTransaction not updating clock line immediately (#9221)
## Description of Change
This PR fixes #9221 by forcing SPI clock line to update within `spiTransaction`.
This is especially important when sharing a single SPI bus between devices with different SPI modes.
Notable use case is using SD card (SPI mode 0) and ST7789-based TFT display (mode 3).

## Tests scenarios
Tested with ESP32-C3 SuperMini board.

## Related links
- #9221
- https://github.com/moononournation/Arduino_GFX/issues/433