espressif/arduino-esp32 #9229: Turning WiFi off with WiFi.disconnect(true) or WiFi.mode(WIFI_OFF) prints error messages even when logging is disabled

emilfihlman, 2024-02-09:
### Board

Sparkfun Thing Plus and ESP32 DevKitC V4

### Device Description

Plain devboard without anything attached

### Hardware Configuration

Nothing attached

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Debian

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Using latest arduino boards manager version 3.0.0-alpha3
Also happened in earlier version

If WiFi.disconnect(true) or WiFi.mode(WIFI_OFF)
E (NUMBER_HERE) wifi:NAN WiFi stop

### Sketch

```cpp
#include <WiFi.h>

const char *ap_ssid="";
const char *ap_key="";

void setup()
{
	Serial.begin(115200);
	Serial.print("setup\n");
	Serial.print("WiFi: mode WIFI_STA\n");
	WiFi.mode(WIFI_STA);
	Serial.print("WiFi: begin\n");
	WiFi.begin(ap_ssid, ap_key);
	Serial.print("WiFi: waiting connect\n");
	while(WiFi.status()!=WL_CONNECTED)
	{
		delay(100);
	}
	Serial.print("WiFi: connected\n");
	Serial.print("WiFi: disconnect\n");
	WiFi.disconnect(false); //Change here to move the error message
	delay(5000);
	Serial.print("WiFi: mode WIFI_OFF\n");
	WiFi.mode(WIFI_OFF);
}

void loop()
{
	Serial.print("loop\n");
	delay(1000);
}
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1256
load:0x40078000,len:13832
load:0x40080400,len:4
load:0x40080404,len:3048
entry 0x40080590
[     3][D][esp32-hal-cpu.c:259] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 100
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   347264 B ( 339.1 KB)
  Free Bytes        :   317308 B ( 309.9 KB)
  Allocated Bytes   :    22972 B (  22.4 KB)
  Minimum Free Bytes:   311796 B ( 304.5 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : DIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  6400.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00650000, size:  6400.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00C90000, size:  3456.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x00FF0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Feb  9 2024 18:33:40
  Compile Host OS   : linux
  ESP-IDF Version   : v5.1.2-185-g3662303f31-dirty
  Arduino Version   : 3.0.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_THING_PLUS
  Arduino Variant   : esp32thing_plus
  Arduino FQBN      : esp32:esp32:esp32thing_plus:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600,DebugLevel=debug,EraseFlash=all
============ Before Setup End ============
setup
WiFi: mode WIFI_STA
[   508][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 0 - WIFI_READY
WiFi: begin
[   583][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 2 - STA_START
WiFi: waiting connect
[   638][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1148][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1156][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 10.100.60.166, MASK: 255.255.192.0, GW: 10.100.0.1
WiFi: connected
WiFi: disconnect
[  1218][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  1227][W][WiFiGeneric.cpp:1064] _eventCallback(): Reason: 8 - ASSOC_LEAVE
WiFi: mode WIFI_OFF
E (12646) wifi:NAN WiFi stop
[  6221][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 3 - STA_STOP
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   347264 B ( 339.1 KB)
  Free Bytes        :   298128 B ( 291.1 KB)
  Allocated Bytes   :    40168 B (  39.2 KB)
  Minimum Free Bytes:   253060 B ( 247.1 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
loop
loop
loop
loop
loop
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-02-10:
@emilfihlman - I'm not able to reproduce the issue using Alpha-3 and the provided example. 
I have changed it to `WiFi.disconnect(true)`, turned on and off debug levels.

Could you please send me more details about how to diplsay the WiFi NAN (Neighbor Awareness Networking) error message?
It is related to this mode: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_nan.html

VojtechBartoska, 2024-09-12:
As we are not able to reproduce and there was no answer, I am closing this ticket.

closed 2024-09-12 by VojtechBartoska