espressif/arduino-esp32 #9231: How to revert to 2.0.14 from 3.0 ??

dagnall53, 2024-02-10:
### Board

ESP32 devboard

### Device Description

ESP32 dev board

### Hardware Configuration

Not applicable

### Version

v2.0.14

### IDE Name

Arduino 2.2.1

### Operating System

win10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Once I tried the update to 3.0.0alpha3, I no longer have the option to revert to 2.0.14.. only 2.0.11 which had errors!

HOW do I get back to 2.0.14 Please - Desperate as 3.0.0 introduces unacceptable changes!
 

### Sketch

```cpp
Not applicable
```

### Debug Message

```plain
Not applicable
```

### Other Steps to Reproduce

Not applicable

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JAndrassy, 2024-02-10:
[image]

JAndrassy, 2024-02-10:
In IDE Preference add the URL to Additional boards URLs

dagnall53, 2024-02-10:
Got it Thanks!.. 
Phew!

closed 2024-02-10 by SuGlider