espressif/arduino-esp32 #9234: HWCDC.cpp  symbols USB_DM_GPIO_NUM, USB_DP_GPIO_NUM nowhere defined

ataweg, 2024-02-11:
### Board

ESP32-S3

### Device Description

--

### Hardware Configuration

--

### Version

latest master (checkout manually)

### IDE Name

ESP-IDF

### Operating System

Windows 11

### Flash frequency

--

### PSRAM enabled

yes

### Upload speed

--

### Description

In file https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HWCDC.cpp
When #if SOC_USB_SERIAL_JTAG_SUPPORTED is valid
then the symbols USB_DM_GPIO_NUM, USB_DP_GPIO_NUM are missing.

These symbols were introduced with commit 6f7a1ca76ada4d705d79ce863450c91349327a79 in Nov 2023

### Sketch

```cpp
--
```

### Debug Message

```plain
--
```

### Other Steps to Reproduce

--

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-02-11:
I can't reproduce any error using the Arduino IDE with the option "USB Mode: Hardware CDC and JTAG" + ESP32-S3 Board.

The symbols are defined in the file `io_mux_reg.h` which is included by 
`#include "soc/io_mux_reg.h"` at
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HWCDC.cpp#L26

Please provide more information about how to reproduce such error, including the error messages.

ataweg, 2024-02-12:
Your are right for branch release/5.1, but in branch release/5.2 and master the symbols have changed their name to USB_INT_PHY0_DM_GPIO_NUM, USB_INT_PHY0_DP_GPIO_NUM.