espressif/arduino-esp32 #9239: GPIO47 and GPIO48 I2C

herry93, 2024-02-13:
### Board

esp32-s3-devkitc-1

### Device Description

DEVKIT C 1

### Hardware Configuration

GPIO 47 and GPIO 48 are connected with I2C

### Version

latest master (checkout manually)

### IDE Name

PatformIO IDE

### Operating System

WIN 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

pins 48 and 47 cannot read I2C properly

### Sketch

```cpp
#include <Arduino.h>
#include "I2C_BM8563.h"

// RTC BM8563 I2C port

// I2C pin definition for M5Stick & M5Stick Plus & M5Stack Core2
#define BM8563_I2C_SDA 48
#define BM8563_I2C_SCL 47

// I2C pin definition for M5Stack TimerCam
// #define BM8563_I2C_SDA 12
// #define BM8563_I2C_SCL 14

I2C_BM8563 rtc(I2C_BM8563_DEFAULT_ADDRESS, Wire1);

void setup() {
  // Init Serial
  Serial.begin(115200);
  delay(50);

  // Init I2C
  Wire1.begin(BM8563_I2C_SDA, BM8563_I2C_SCL);

  // Init RTC
  rtc.begin();
}

void loop() {
  I2C_BM8563_DateTypeDef dateStruct;
  I2C_BM8563_TimeTypeDef timeStruct;

  // Get RTC
  rtc.getDate(&dateStruct);
  rtc.getTime(&timeStruct);

  // Print RTC
  Serial.printf("%04d/%02d/%02d %02d:%02d:%02d\n",
                dateStruct.year,
                dateStruct.month,
                dateStruct.date,
                timeStruct.hours,
                timeStruct.minutes,
                timeStruct.seconds
               );

  // Wait
  delay(1000);
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x158
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa88
load:0x403cc700,len:0x2800
SHA-256 comparison failed:
Calculated: 1131788a3f6eb2d0c83fe2b5114a644b81ce26ee911a2e3e1457bcf0848203d1
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c9868
2000/00/00 00:00:00
2000/00/00 00:00:00
2000/00/00 00:00:00
2000/00/00 00:00:00
2000/00/00 00:00:00
2000/00/00 00:00:00
2000/00/00 00:00:00
2000/00/00 00:00:00
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-02-13:
@me-no-dev - Please take a look. Thanks.

SuGlider, 2024-02-13:
@herry93 - I know that some ESP32-S3 devkit use IO48 as RGB LED. 
This is the case for the M1 devkit, but should not be the case for the C1.

Can you just check if the board LED is IO38 or IO48? Thanks!

SuGlider, 2024-02-13:
@herry93 - If you use other pins for I2C, does it work fine?

herry93, 2024-02-13:
> @herry93- Saya tahu bahwa beberapa devkit ESP32-S3 menggunakan IO48 sebagai LED RGB. Hal ini berlaku pada devkit M1, namun tidak berlaku pada C1.
> 
> Bisakah Anda memeriksa apakah LED papannya IO38 atau IO48? Terima kasih!

The PIN for the LED is on GPIO48

pin 48 can't just blink

herry93, 2024-02-13:
> @herry93- Jika Anda menggunakan pin lain untuk I2C, apakah berfungsi dengan baik?

Yes, of course I tried using another GPIO and it worked normally

TD-er, 2024-02-13:
Not sure where (or when) I saw this, but if I'm not mistaken I have seen some peripheral code recently where pin masks were stored per 24 bit in 32-bit words as some RTOS shared values.
As pin 48 is outside this range of 2x 24-bit, could this be related?

Sorry I really can't remember where I saw this.
I only remember thinking "hmm that's odd" when I saw it.

TD-er, 2024-02-13:
I finally found an S3-board which has GPIO-47 & 48 exposed and hooked it up to a logic analyzer.

At least it seems to be able to send data (have not yet hooked up some I2C device to it)

[image]

SDA => GPIO-47
SCL => GPIO-48

me-no-dev, 2024-02-13:
@herry93 do you have pullups on those pins?

SuGlider, 2024-02-13:
> > @herry93- Saya tahu bahwa beberapa devkit ESP32-S3 menggunakan IO48 sebagai LED RGB. Hal ini berlaku pada devkit M1, namun tidak berlaku pada C1.
> > Bisakah Anda memeriksa apakah LED papannya IO38 atau IO48? Terima kasih!
> 
> The PIN for the LED is on GPIO48

Consider that there is a WS2812b RGB LED connected to this pin.
Therefore there is circuit conneting IO48 to GND passing by the resistor and the LED microcontroller that is causing the issue.

> 
> pin 48 can't just blink

This is not a regular LED that would light up just by setting the IO to 3.3V. There is a protocol to make it work.
I think that it may lack, as @me-no-dev said, some pull up to hold the line HIGH.

herry93, 2024-02-15:
> Saya akhirnya menemukan papan S3 yang telah mengekspos GPIO-47 & 48 dan menghubungkannya ke penganalisis logika.
> 
> Setidaknya tampaknya dapat mengirim data (belum menghubungkan beberapa perangkat I2C ke sana)
> 
> ![gambar](https://private-user-images.githubusercontent.com/3751318/304412593-3bb26bea-ddbe-41ce-9286-ded8f58ae0a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5NjEyMjMsIm5iZiI6MTcwNzk2MDkyMywicGF0aCI6Ii8zNzUxMzE4LzMwNDQxMjU5My0zYmIyNmJlYS1kZGJlLTQxY2UtOTI4Ni1kZWQ4ZjU4YWUwYTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTVUMDEzNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWZjZDEzMjRmNzZjMWUyNzQxNWJlYzQ3YzM2ZmFkZTNiYjRkYzI1NWY1NzI3NjEyNDkwZWM2NWVmZWM0YjViYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.k_6RSCfpGaH9NKNLUdZAzKVBlphLD3KIphijkUhEkr8)
> 
> SDA => GPIO-47 SCL => GPIO-48
> 
> Sunting: Menukar SDA/SCL juga tampaknya mengirimkan nilai I2C yang tepat: ![gambar](https://private-user-images.githubusercontent.com/3751318/304413216-864b46ab-0c64-482f-b67c-af79b631e11b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5NjEyMjMsIm5iZiI6MTcwNzk2MDkyMywicGF0aCI6Ii8zNzUxMzE4LzMwNDQxMzIxNi04NjRiNDZhYi0wYzY0LTQ4MmYtYjY3Yy1hZjc5YjYzMWUxMWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTVUMDEzNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Nzk5NjRhOTZkNzRiMGI2MmMwM2FkZTJhYTU5ZGIwMDYwNGQ3ZDdiNjI0MzUwMGYwMDVmNWVlYjA2MjRkMjAwMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.tDyLRpQk8ArLgqnn-Y62mpI8qJO4apV4hy4AZiIz61o)

where did you find this?
and what S3 Board are you using?

herry93, 2024-02-15:
> > > @herry93- Saya tahu bahwa beberapa devkit ESP32-S3 menggunakan IO48 sebagai LED RGB. Hal ini berlaku pada devkit M1, namun tidak berlaku pada C1.
> > > Bisakah Anda memeriksa apakah LED papannya IO38 atau IO48? Terima kasih!
> > 
> > 
> > PIN untuk LED ada di GPIO48
> 
> Pertimbangkan ada LED RGB WS2812b yang terhubung ke pin ini. Oleh karena itu ada sirkuit yang menghubungkan IO48 ke GND yang melewati resistor dan mikrokontroler LED yang menyebabkan masalah.
> 
> > pin 48 tidak bisa berkedip begitu saja
> 
> Ini bukan LED biasa yang akan menyala hanya dengan mengatur IO ke 3.3V. Ada protokol untuk membuatnya berfungsi. Saya pikir itu mungkin kurang, misalnya@me-no-devkatanya, ada yang pull up untuk menahan garis TINGGI.

I have used PULLUP and I have removed the WS2812 LED.

herry93, 2024-02-15:
> Pertimbangkan ada LED RGB WS2812b yang terhubung ke pin ini. Oleh karena itu ada sirkuit yang menghubungkan IO48 ke GND yang melewati resistor dan mikrokontroler LED yang menyebabkan masalah.
>
>
> Ini [...]
I have used PULLUP and I have removed the WS2812 LED.

TD-er, 2024-02-15:
> > I finally found an S3 board that had GPIO-47 & 48 exposed and connected it to a logic analyzer.
> > At least it seems to be able to send data (haven't connected some I2C devices to it yet)
> > ![picture](https://private-user-images.githubusercontent.com/3751318/304412593-3bb26bea-ddbe-41ce-9286-ded8f58ae0a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5NjEyMjMsIm5iZiI6MTcwNzk2MDkyMywicGF0aCI6Ii8zNzUxMzE4LzMwNDQxMjU5My0zYmIyNmJlYS1kZGJlLTQxY2UtOTI4Ni1kZWQ4ZjU4YWUwYTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTVUMDEzNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWZjZDEzMjRmNzZjMWUyNzQxNWJlYzQ3YzM2ZmFkZTNiYjRkYzI1NWY1NzI3NjEyNDkwZWM2NWVmZWM0YjViYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.k_6RSCfpGaH9NKNLUdZAzKVBlphLD3KIphijkUhEkr8)
> > SDA => GPIO-47 SCL => GPIO-48
> > Edit: Swapping SDA/SCL also seems to send the correct I2C values:![picture](https://private-user-images.githubusercontent.com/3751318/304413216-864b46ab-0c64-482f-b67c-af79b631e11b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5NjEyMjMsIm5iZiI6MTcwNzk2MDkyMywicGF0aCI6Ii8zNzUxMzE4LzMwNDQxMzIxNi04NjRiNDZhYi0wYzY0LTQ4MmYtYjY3Yy1hZjc5YjYzMWUxMWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTVUMDEzNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Nzk5NjRhOTZkNzRiMGI2MmMwM2FkZTJhYTU5ZGIwMDYwNGQ3ZDdiNjI0MzUwMGYwMDVmNWVlYjA2MjRkMjAwMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.tDyLRpQk8ArLgqnn-Y62mpI8qJO4apV4hy4AZiIz61o)
> 
> where did you find this? and what S3 Board are you using?

I have several LoLin S3 boards
I think it is [this one (the S3 N16R8)](https://www.aliexpress.com/item/1005004609461716.html)
And I just used my own firmware (ESPEasy) and performed an I2C scan.
Then captured the signals using my Logic Analyzer.

me-no-dev, 2024-02-15:
I2C working fine on 47(SDA) and 48(SCL) for me

closed 2025-01-15 by Parsaabasi