espressif/arduino-esp32 #9240: Creating a Thread/Matter device with ESP32 C6 and Arduino Core

BorisBrock, 2024-02-13:
### Related area

Smart Home

### Hardware specification

ESP32 C6

### Is your feature request related to a problem?

I would like to create some simple Thread/Matter devices for my smart home. A Open Thread Border Router (based on Raspberry Pi) and a Home Assistant instance (with Matter server) are already running.

So I purchased a few ESP32 C6 DevKits (because of their Thread support). But I cannot find any good examples in this repository on how to create simple custom Thread/Matter devices with the ESP32 ARduino Core.

### Describe the solution you'd like

Could you provide a basic example to show how Thread/Matter devices can be created with the ESP32 C6 and the Arduino Core?

### Describe alternatives you've considered

Using Espressif ZeroCode is no option, since the ESP32 C6 seems not to be supported (why???). Using the Espressif IDF would be another option, but seems to be much more complicated than using the Arduino Core.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-02-13:
OpenThread is not currently compiled into the arduino-esp32 stack.

SuGlider, 2024-02-13:
@BorisBrock - We are currently working on integrating OpenThread into Arduino. 
After finishing it, the next step will be integrate Matter.
It may take some time, but it will be done along this year.

In the meantime a possible solution would be to use Arduino as IDF Component and integrate Thread and Matter into the project. The available examples are in the IDF Github. 
There are just basic examples about Thread and many about Matter.

https://github.com/espressif/esp-idf/tree/master/examples/openthread
https://github.com/espressif/openthread/tree/main/examples

https://github.com/espressif/esp-matter/tree/main/examples
https://github.com/espressif/connectedhomeip/tree/test_event_4/examples

VojtechBartoska, 2024-02-14:
Hello @BorisBrock, we will use this ticker for progress tracking.

VojtechBartoska, 2024-02-20:
Update:

There is older ticket with more info, I am closing this one in favor of https://github.com/espressif/arduino-esp32/issues/7432