espressif/arduino-esp32 #9244: BluetoothSerial: register_callback compile error

ericlangel, 2024-02-13:
### Board

ESP32 Dev Module

### Device Description

DevKitC

### Hardware Configuration

no

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows

### Flash frequency

-

### PSRAM enabled

no

### Upload speed

-

### Description

same Problem as #7675 but with actual 3.0.0 alpha

### Sketch

```cpp
#include "BluetoothSerial.h"
 
BluetoothSerial SerialBT;
 
void callback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param){
  if(event == ESP_SPP_SRV_OPEN_EVT){
    Serial.println("Client Connected");
  }
 
  if(event == ESP_SPP_CLOSE_EVT ){
    Serial.println("Client disconnected");
  }
}
 
void setup() {
  Serial.begin(115200);
 
  SerialBT.register_callback(callback);
 
  if(!SerialBT.begin("ESP32")){
    Serial.println("An error occurred initializing Bluetooth");
  }else{
    Serial.println("Bluetooth initialized");
  }
}
 
void loop() {}
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-02-13:
@ericlangel Thanks for reporting! PR #9245 shall solve it.

ericlangel, 2024-02-14:
yes, it worked

PR #9245 (2024-02-14): Fixes Pointer and Call using Serial BT callback
## Description of Change
IDF has changed Serial BT Callback type definition. This PR fixes the Arduino SerialBT API to match new type and function signature.

## Tests scenarios
ESP32 - CI

``` cpp
#include "BluetoothSerial.h"
 
BluetoothSerial SerialBT;
 
void callback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param){
  if(event == ESP_SPP_SRV_OPEN_EVT){
    Serial.println("Client Connected");
  }
 
  if(event == ESP_SPP_CLOSE_EVT ){
    Serial.println("Client disconnected");
  }
}
 
void setup() {
  Serial.begin(115200);
 
  SerialBT.regist [... cut]