espressif/arduino-esp32 #9246: [W5500] M5Stack AtomPoE doesn't have INT/RST connected

TD-er, 2024-02-13:
### Board

M5Stack AtomS3 + ATOM_PoE

### Device Description

See [schematic of AtomPoE](https://docs.m5stack.com/en/atom/atom_poe)
N.B. the pins on the AtomPoE board refer to an ESP32 Atom module.
The AtomS3 uses an ESP32-S3 and thus has different GPIO pins.

### Hardware Configuration

SPI CLK: GPIO-5
SPI MISO: GPIO-7
SPI MOSI: GPIO-8
W5500 CS: GPIO-6

INT & RST  set to -1

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

There are boards with a W5500 SPI Ethernet controller which do not have the INT and RST pin connected.
However in the `ETH::beginSPI` function, this will be rejected as it checks for those pins.

To overcome this check, this function can be changed to something like this:
```c++
bool ETHClass::beginSPI(eth_phy_type_t type, uint8_t phy_addr, int cs, int irq, int rst, 
#if ETH_SPI_SUPPORTS_CUSTOM
    SPIClass *spi, 
#endif
    int sck, int miso, int mosi, spi_host_device_t spi_host, uint8_t spi_freq_mhz){
    esp_err_t ret = ESP_OK;

    if(_eth_started || _esp_netif != NULL || _eth_handle != NULL){
        log_w("ETH Already Started");
        return true;
    }
    if(cs < 0 || 
	   (
#if CONFIG_ETH_SPI_ETHERNET_W5500
        (ETH_PHY_W5500 != type) &&
#endif
		irq < 0)){
        log_e("CS and IRQ pins must be defined!");
        return false;
    }
    ...
```

However even with those changes, I can't get the `ETH.begin` function to return true.
N.B. I do call the ETH.begin function with `SPI` as argument. 
On M5Stack Core2 (ESP32) with W5500 base plate and on ESP32-C3 board wih DM9051 this does work perfectly fine.
But those setups have INT and RST wired to the ESP.

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2024-02-14:
Is there already some work-in-progress publicly visible?

TD-er, 2024-02-15:
> This will be soon supported. IDF team is working on it.

Already made a PR for it :)

closed 2024-02-22 by me-no-dev

MeisterQ, 2024-05-18:
Is here any information how to use this in Arduino IDE? i have a board without IRQ and RST connected. How to use it in Arduino IDE?