espressif/arduino-esp32 #9248: SPI peripheral / bus numbers on ESP32S3 wrong and overly limited

anm, 2024-02-13:
### Board

Any ESP32-S3

### Device Description

n/a

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

n/a

### Operating System

n/a

### Flash frequency

n/a

### PSRAM enabled

yes

### Upload speed

n/a

### Description

The ESP32S3 datasheet says it has four SPI controllers, numbered 0, 1, 2, and 3.

`esp32-hal-spi.c` says:
`#define SPI_COUNT           (2)`

This causes it to reject starting the peripherals numbered 2 and 3.

`esp32-hal-spi.h` also defines only:
```
#define FSPI  0
#define HSPI  1
```

Related issue for ESP32-C3:
https://github.com/espressif/arduino-esp32/issues/7596

### Sketch

```cpp
e.g.: new SPIClass(3)
```

### Debug Message

```plain
None, but there should be proper error reporting.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-02-14:
SPI0 and SPI1 are to be used only for Flash and PSRAM. SPI2 and SPI3 are available for use. It is a bit confusing, but you do have two buses only and they do not start with 0

me-no-dev, 2024-02-14:
We will be "fixing this" in some way in a near future to make it more convenient for users

VojtechBartoska, 2024-02-20:
Closing this in favour of https://github.com/espressif/arduino-esp32/issues/8192, we will track this there.