espressif/arduino-esp32 #9271: An error occurred while compiling the Esp32 project using Arduino. Switching to a 2.0.11 \ 2.0.12 \ 2.0.14 \ 3.0.0 Alpha version library and multiple IDE versions reported an error of "\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory"

BalancedTernary, 2024-02-21:
### Board

ESP32S3 Dev Module

### Device Description

ESP32S3 Dev Module

### Hardware Configuration

not anything

### Version

v2.0.14

### IDE Name

Arduino(Multiple versions, including the latest version)

### Operating System

Windows10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Any development board that involves ESP32 will compile incorrectly, while a development board that does not involve ESP32 can be successfully compiled

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
FQBN: esp32:esp32:esp32s3
使用平台的 ‘esp32s3’ 开发板，在列出的文件夹中：C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11
使用平台的 ‘esp32’ 代码，在列出的文件夹中：C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11

cmd /c if exist "D:\UserName\Downloads\ESP32-S3-Touch-LCD-4.3_Code\Arduino\lvgl_Porting\partitions.csv" COPY /y "D:\UserName\Downloads\ESP32-S3-Touch-LCD-4.3_Code\Arduino\lvgl_Porting\partitions.csv" "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\partitions.csv"
cmd /c if not exist "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\partitions.csv" if exist "C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\variants\esp32s3\partitions.csv" COPY "C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\variants\esp32s3\partitions.csv" "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\partitions.csv"
cmd /c if not exist "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\partitions.csv" COPY "C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\tools\partitions\default.csv" "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\partitions.csv"
cmd /c IF EXIST "D:\UserName\Downloads\ESP32-S3-Touch-LCD-4.3_Code\Arduino\lvgl_Porting\bootloader.bin" ( COPY /y "D:\UserName\Downloads\ESP32-S3-Touch-LCD-4.3_Code\Arduino\lvgl_Porting\bootloader.bin" "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\lvgl_Porting.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\variants\esp32s3\bootloader.bin" ( COPY "C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\variants\esp32s3\bootloader.bin" "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\lvgl_Porting.ino.bootloader.bin" ) ELSE ( "C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\lvgl_Porting.ino.bootloader.bin" "C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\tools\sdk\esp32s3\bin\bootloader_qio_80m.elf" ) )
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
cmd /c if exist "D:\UserName\Downloads\ESP32-S3-Touch-LCD-4.3_Code\Arduino\lvgl_Porting\build_opt.h" COPY /y "D:\UserName\Downloads\ESP32-S3-Touch-LCD-4.3_Code\Arduino\lvgl_Porting\build_opt.h" "D:\TEMP\arduino\sketches\02A85770975DD838CA2F7B97F8440F71\build_opt.h"
[... 3 lines cut ...]
C:\Users\UserName\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32s3-elf-g++ -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.5" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/newlib/platform_include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/freertos/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/freertos/include/esp_additions/freertos -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/freertos/port/xtensa/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/freertos/include/esp_additions -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_hw_support/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_hw_support/include/soc -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_hw_support/include/soc/esp32s3 -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3 -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3/private_include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/heap/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/log/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/lwip/include/apps -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/lwip/include/apps/sntp -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/lwip/lwip/src/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/lwip/port/esp32/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/lwip/port/esp32/include/arch -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/soc/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/soc/esp32s3 -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/soc/esp32s3/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/hal/esp32s3/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/hal/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/hal/platform_port/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_rom/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_rom/include/esp32s3 -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_rom/esp32s3 -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_common/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_system/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_system/port/soc -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_system/port/public_compat -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/xtensa/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/xtensa/esp32s3/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/driver/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/driver/esp32s3/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_pm/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_ringbuf/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/efuse/include -IC:\Users\UserName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include [... cut ...]
c:\users\UserName\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory
#include <bits/c++allocator.h> // Define the base class to std::allocator.
^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-02-21:
As far as I remember, this issue is caused by path to the file being too long. @SuGlider @lucasssvaz do you remember if there was any fix for Windows or reason that max path length to be different for some reason?

@BalancedTernary what is the file system of the C drive?

BalancedTernary, 2024-02-21:
> As far as I remember, this issue is caused by path to the file being too long. @SuGlider @lucasssvaz do you remember if there was any fix for Windows or reason that max path length to be different for some reason?
> 
> @BalancedTernary what is the file system of the C drive?

The file system of my c drive is NTFS.

Jason2866, 2024-02-21:
It makes no difference which Filesystem is used. The max. len for a cmd is 8191 bytes when using Windows.

me-no-dev, 2024-02-21:
@Jason2866 it's not command length, but path length, which was something like 256, but was often found to be a bit less. We already had that issue with RV toolchain and 3.0.0, where we had to rename the package in order to save some bytes for longer usernames.

BalancedTernary, 2024-02-21:
> @Jason2866 it's not command length, but path length, which was something like 256, but was often found to be a bit less. We already had that issue with RV toolchain and 3.0.0, where we had to rename the package in order to save some bytes for longer usernames.

I don't know if this information will be useful: I looked for the "c++allocator.h" file in the "/bits" directory, and it was indeed not found. It looks like it includes a file that doesn't exist, or is in the wrong location.

Jason2866, 2024-02-21:
@me-no-dev Oh, my. Didn't know the issue about Path length. CMD length is restricted as I wrote too. This "bites" in Platformio using Windows. https://github.com/platformio/platformio-core/pull/4856

lucasssvaz, 2024-02-21:
@me-no-dev I remember that this could happen if the path was too long or the extraction of the libraries failed somehow.
I checked our talks regarding this and the path limit is 200 characters. It was fixed before in #8750.

@BalancedTernary I would recommend running the get script again to make sure everything was extracted properly.

BalancedTernary, 2024-02-21:
> @BalancedTernary if you can not find the file anywhere in the toolchain folder, then because of that same issue, it was not even extracted.

I can find these files under the toolchain folder, but they don't find the directory indicated in the error log. It actually exists in "AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\xtensa-esp32s3-elf\bits" , but gcc is found in "AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\bits"

SuGlider, 2024-02-21:
It really seems like the combination of a 23 characters username and the long path is preventing Windows tools from being able to open the files.

The issue seems related to the number of characters from a PATH that Windows is able to deal in GCC tools.

A Not perfect solution could be trying to install the Arduino under a "short user name" and check if this makes it work.

SuGlider, 2024-02-21:
I've found a Windows Microsoft discussion about a problem with path longer than 256 characters:
https://developercommunity.visualstudio.com/t/compiler-cant-find-source-file-in-path/10221576

It sounds like a general problem with Windows... Not sure what we could do to help it, beyond what has been done in #8750

Jason2866, 2024-02-21:
The currently only real solution is not to use Windows. Using a VM running Ubuntu and everything just works.

BalancedTernary, 2024-02-22:
> It really seems like the combination of a 23 characters username and the long path is preventing Windows tools from being able to open the files.
>
> The issue seems related to the number of characters [...]

I tried to use a short username, but the same error still occurred. I tried to copy all the files in "AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\xtensa-esp32s3-elf\bits" to "AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\bits", and now the problem is temporarily solved.

SuGlider, 2024-02-26:
@BalancedTernary - Can you please check this Windows registry: `Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem`

It should be set as 1 (true). When it is '0', Windows will limit the path to 260 chars.
Setting it to '1' will extend this limit to 32,767 chars.

https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry#enable-long-paths-in-windows-10-version-1607-and-later

Go to the command search text box at the TaskBar, run "regedit.exe" and browse the Registry directory up to 
`Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem`
Double click `FileSystem` and check/change its value.

VojtechBartoska, 2024-09-12:
Hello @BalancedTernary, can this ticket be closed as answered?

closed 2025-01-15 by Parsaabasi