espressif/arduino-esp32 #9274: ESP32-S3 and SDMMC not working

iharshadec, 2024-02-21:
### Board

ESP32-S3 Dev Kit

### Device Description

ESP32-S3 Dev kit with Breadboard and SD Socket 

### Hardware Configuration

int clk = 14;
int cmd = 15;
int d0 = 2;
int d1 = 4;
int d2 = 12;
int d3 = 13;

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

NA

### PSRAM enabled

no

### Upload speed

921600

### Description

I am Trying to get SDIO interface working with ESP32-S3 and SD Card.
This is the example code I am trying:- https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC

Modifications I have done is I have changed Pins.

```
//! Changing this Pins
//  int clk = 36;
//  int cmd = 35;
//  int d0 = 37;
//  int d1 = 38;
//  int d2 = 33;
//  int d3 = 39; // GPIO 34 is not broken-out on ESP32-S3-DevKitC-1 v1.1

int clk = 14;
int cmd = 15;
int d0 = 2;
int d1 = 4;
int d2 = 12;
int d3 = 13;
```

### Sketch

```cpp
/*
 * pin 1 - D2                |  Micro SD card     |
[... 341 lines cut ...]
    readFile(SD_MMC, "/hello.txt");
    deleteFile(SD_MMC, "/foo.txt");
    renameFile(SD_MMC, "/hello.txt", "/foo.txt");
    readFile(SD_MMC, "/foo.txt");
    testFileIO(SD_MMC, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD_MMC.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
}

void loop()
{
}
```

### Debug Message

```plain
E (6009) sdmmc_io: sdmmc_io_reset: unexpected return: 0x108
E (6009) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
Card Mount Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-02-22:
Hello @iharshadec, can you please set Debug level to "Verbose" and post whole log from Serial Monitor? Thanks

iharshadec, 2024-02-22:
> Hello @iharshadec, can you please set Debug level to "Verbose" and post whole log from Serial Monitor? Thanks
> 
> Also you are trying to use all 4 data pins, but in SDMMC.begin() you are setting 1bitmode to true. Its the 2nd parameter.
> 
> ```
> if (!SD_MMC.begin("/sdcard", true, false, 5000, 5))
>     {
>         Serial.println("Card Mount Failed");
>         return;
>     }
> ```

Hi @P-R-O-C-H-Y 
Thanks for pointing out this. I missed to change this back to false as I am using all pins and 4-bit mode.

This time I do see files being created but still at last there is some error I see

```
SD_MMC Card Type: SDHC
SD_MMC Card Size: 3840MBListing directory: /  DIR : System Volume Information
Creating Dir: /mydirDir created
Listing directory: /  DIR : System Volume Information
  DIR : mydir
Removing Dir: /mydirDir removed
Listing directory: /  DIR : System Volume Information
Listing directory: /System Volume Information  FILE: WPSettings.dat  SIZE: 12
  FILE: IndexerVolumeGuid  SIZE: 76
Writing file: /hello.txtFile written
Appending to file: /hello.txtMessage appended
Reading file: /hello.txtRead from file: Hello World!Deleting file: /foo.txt[  7832][E][vfs_api.cpp:182] remove(): /foo.txt does not exists or is directory
Delete failed
Renaming file /hello.txt to /foo.txtFile renamed
Reading file: /foo.txtRead from file: Hello World![  7877][E][vfs_api.cpp:105] open(): /sdcard/test.txt does not exist, no permits for creation
Failed to open file for reading
1048576 bytes written for 7462 msTotal space: 3832MBUsed space: 1MB
```

me-no-dev, 2024-02-22:
@iharshadec please take care to properly paste your logs!

From what I see, everything is fine. first error is because you want to remove `foo.txt`, but you have `hello.txt` created, then second error is you want to open `test.txt`, but you have `foo.txt` only. Everything is fine.

closed 2024-02-22 by iharshadec