espressif/arduino-esp32 #9276: ESP32-S3 USB HOST PENDRIVE INTERFACE

micromagine, 2024-02-22:
### Related area

USB HOST MSC

### Hardware specification

ESP32-S3, ESP32-S2

### Is your feature request related to a problem?

I WOULD TRIED ALL TINY-USB AND USB MSC CODE FOR DEVELOPMENT BUT I CANNOT FIND ANY CODE FOR USB MSC HOST.

### Describe the solution you'd like

https://github.com/nathalislight/esp32s3_USB_Drive/tree/main

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

micromagine, 2024-03-04:
please try this code

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

The3ven, 2025-07-16:
please bring this to life i desperately need this in arduino ide examples for esp32s3, i would also like to bring your attenetion on this example someone port from idf https://github.com/esp32beans/ESP32_USB_Host_HID please bring them to arduino