espressif/arduino-esp32 #9278: following docs/en/lib_builder.rst does not produce a working sdk

folkertvanheusden, 2024-02-22:
### Board

ESP*

### Device Description

not relevant

### Hardware Configuration

not relevant

### Version

latest master (checkout manually)

### IDE Name

platformio

### Operating System

linux

### Flash frequency

123

### PSRAM enabled

yes

### Upload speed

115200

### Description

When running `./build.sh` and copy the result from `out/tools/esp32-arduino-libs/` to `.platformio/packages/framework-arduinoespressif32/tools/sdk/` then even the simplest Arduino-application won't compile due to stuff missing:
e.g.

```
void setup() {
	Serial.begin(115200);
}

void loop() {
	Serial.println(F("Hello"));
}
```

produces:

```cpp
Compiling .pio/build/ESP32-wemos/src/test.ino.cpp.o
In file included from /home/folkert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/compare_set.h:12,
                 from /home/folkert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:13,
                 from /home/folkert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:42,
                 from /home/folkert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/portable.h:51,
                 from /home/folkert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:63,
                 from /home/folkert/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from /tmp/tmpw9drajss:1:
/home/folkert/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/soc_memory_types.h:9:2: warning: #warning "soc_memory_types.h is deprecated, please migrate to esp_memory_utils.h" [-Wcpp]
 #warning "soc_memory_types.h is deprecated, please migrate to esp_memory_utils.h"
  ^~~~~~~
...
         rmt_config_t config = RMT_DEFAULT_ARD_CONFIG_TX(pin, channel, buffers);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/folkert/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:58:27: error: 'RMT_IDLE_LEVEL_LOW' undeclared (first use in this function)
             .idle_level = RMT_IDLE_LEVEL_LOW,        \
                           ^~~~~~~~~~~~~~~~~~
/home/folkert/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:611:31: note: in expansion of macro 'RMT_DEFAULT_ARD_CONFIG_TX'
         rmt_config_t config = RMT_DEFAULT_ARD_CONFIG_TX(pin, channel, buffers);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/folkert/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:614:28: warning: implicit declaration of function 'rmt_driver_install' [-Wimplicit-function-declaration]
             esp_err_code = rmt_driver_install(channel, 0, 0);
                            ^~~~~~~~~~~~~~~~~~
/home/folkert/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:68:21: error: 'RMT_MODE_RX' undeclared (first use in this function); did you mean 'RMT_MEM_64'?
         .rmt_mode = RMT_MODE_RX,                \
                     ^~~~~~~~~~~
/home/folkert/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:617:31: note: in expansion of macro 'RMT_DEFAULT_ARD_CONFIG_RX'
         rmt_config_t config = RMT_DEFAULT_ARD_CONFIG_RX(pin, channel, buffers);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/folkert/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:622:58: error: 'RMT_MEM_OWNER_RX' undeclared (first use in this function); did you mean 'RMT_MEM_ITEM_NUM'?
             esp_err_code = rmt_set_memory_owner(channel, RMT_MEM_OWNER_RX);
                                                          ^~~~~~~~~~~~~~~~
                                                          RMT_MEM_ITEM_NUM
*** [.pio/build/ESP32-wemos/FrameworkArduino/esp32-hal-rmt.c.o] Error 1
```

### Sketch

```cpp
void setup() {
	Serial.begin(115200);
}

void loop() {
	Serial.println(F("Hello"));
}
```

### Debug Message

```plain
not applicable
```

### Other Steps to Reproduce

./build.sh

I did add `CONFIG_SPIRAM_SPEED_40M=yCONFIG_ETH_TRANSMIT_MUTEX=y` to `configs/defconfig.*` but without; that same problem.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-02-22:
you are probably building for the wrong version. The `release/v4.4` branch in the lib-builder will build for Arduino 2.0.x (not sure the PIO version) and master will build for Arduino 3.0.0, which PIO does not yet support.

folkertvanheusden, 2024-02-22:
Hi,
Do you mean `./build.sh -I release/v4.4`? Because that gives the same problems.

folkertvanheusden, 2024-02-22:
Trying that: but it fails. I first did a clean git clone of the repo and then ran
`./build.sh -I release/v4.4 -t esp32 configs/defconfig.esp32`

This fails with:

```
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: espressif/esp-sr
.....NOTICE: Skipping optional dependency: espressif/esp-sr
HINT: Please check manifest file of the following component(s): main
CMake Error at esp-idf/tools/cmake/build.cmake:463 (message):
  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  ERROR: Because project depends on idf (>=5.1) which doesn't match any
  versions, version solving failed.

Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:6 (project)
```

Jason2866, 2024-02-22:
You are using Arduino branch `master` in your last try. Thats the wrong Arduino branch.
You have to use Arduino branch `release/v2.x`

me-no-dev, 2024-02-22:
```bash
git checkout release/v4.4 && ./build.sh
```

Jason2866, 2024-02-22:
All commands:
```
# Install packages, clone repo
$ git clone git@github.com:espressif/esp32-arduino-lib-builder.git -b release/v4.4
$ cd esp32-arduino-lib-builder 
$ ./build.sh -A release/v2.x -I release/v4.4 -t esp32
```

folkertvanheusden, 2024-02-22:
That works!
Thanks for your help.

closed 2024-02-22 by folkertvanheusden